openhwfoundation/core-v-verif #1060: Question about PMA region configs

MikeOpenHWGroup, 2021-12-09:
When looking at pull request #1059 I realized that my understanding of the definition of PMA configurations for the E40X/S cores supported by core-v-verif was incomplete.  The purpose of this issue is determine if I understand things correctly.  There is a question at the end, but it requires some preamble...

In [uvmt_cv32e40x_constansts.sv](https://github.com/openhwgroup/core-v-verif/blob/master/cv32e40x/tb/uvmt/uvmt_cv32e40x_constants.sv#L91) there is a set of parameter definitions for the core's PMA_CFG[] top-level parameter.  In each of these definitions the 32-bit values of `word_addr_high` and `word_addr_low` are shifted right 2 places.
```
'{word_addr_low : 32'h2000_0000>>2, word_addr_high : 32'h3FFF_FFFF>>2, main : 1'b1, bufferable : 1'b1, cacheable : 1'b0, atomic : 1'b1}
```
It appears that the PMA configs the values of `word_addr_high` and `word_addr_low` are written to make them match what a human reader would expect when thinking about a 32-bit flat address space.  For example, [PMA_DEBUG_CFG](https://github.com/openhwgroup/core-v-verif/blob/master/cv32e40x/tb/uvmt/uvmt_cv32e40x_constants.sv#L70) has an entry that looks like this:
```
word_addr_low  : ('h 1a11_0800 - 'd 16) >> 2,
word_addr_high : 'h 1a11_0800 >> 2,
```
The above defines a PMA region of 16 bytes starting at `32'h1a11_0800`, which the default value of `dm_halt_addr` in the CV32E40X environment.

PMA regions sit inside a 34-bit address space that is word-aligned (addr[1:0]==2'b00).  According to the [User Manual](https://docs.openhwgroup.org/projects/cv32e40x-user-manual/pma.html#address-range), word_addr_low/word_addr_high are mapped to bits [33:2] of a 34-bit, word aligned address, and in the case of CV32E40X, addr[33:32] is always 2'b00.  Another way to conceptualize that could be:
```
addr_boundary_low[33:0] =  {2'b00, word_addr_low[31:2],  2'b00}
addr_boundary_high[33:0] = {2'b00, word_addr_high[31:2], 2'b00}
```
So here is my question: is the above correct?  That is, the right shifts in the definition of PMA_CFGs are there to hide the fact that word_addr[31:0] map to bits 33:2 of 34-bit address and present readable 32-bit address to the human reader of the code.

silabs-hfegran, 2021-12-09:
Hi @MikeOpenHWGroup 

> It appears that the PMA configs the values of word_addr_high and word_addr_low are written to make them match what a 
> human reader would expect when thinking about a 32-bit flat address space. 

Yes, this is the intention of writing the byte-address + shift rather than the word address itself. 

>>addr_boundary_low[33:0] =  {2'b00, word_addr_low[31:2],  2'b00}
>>addr_boundary_high[33:0] = {2'b00, word_addr_high[31:2], 2'b00}

>So here is my question: is the above correct? That is, the right shifts in the definition of PMA_CFGs are there to hide the fact that word_addr[31:0] map to bits 33:2 of 34-bit address and present readable 32-bit address to the human reader of the code.

Yes, that will be the correct assumption - the upper two bits are zero-extended when we do not support virtual memory.
For reference, a similar question, with somewhat different verbiage, was also raised here: https://github.com/riscv/riscv-isa-manual/issues/131

MikeOpenHWGroup, 2021-12-09:
Thanks for the quick answer and the pointer to the ISA Manual issue @silabs-hfegran.  Much appreciated.