openhwfoundation/core-v-verif #1078: Compilation of test program for cv32e40p and verilator

mohamedElbouazzati, 2021-12-17:
## Bug : compiling the test program
I'm trying to reproduce tests for cv32e40p using verilator simulator and i found this error( can't find -lcv-verif) during compiling helloworld test program, i tried different toolchains from eembecosm but still no results 

[image]

### Type

`/opt/riscv/bin/../lib/gcc/riscv32-unknown-elf/12.0.0/../../../../riscv32-unknown-elf/bin/ld: cannot find -lcv-verif`

### Steps to Reproduce
1. Clone this repo
2. Install verilator from GitHub 
3. Checked with three different  TC from https://www.embecosm.com/resources/tool-chain-downloads/ 
4`. cd /core-verif/cv32e40p/sim/core`
5. `make 
`

Any help please 

Thank you so much

MikeOpenHWGroup, 2021-12-17:
Thanks for the detailed issue report @mohamedElbouazzati, it really helps.

It appears you toolchain is missing the `cv-verif` library.  Can you confirm which version of the toolchain you are using?  Quite honestly, I am not a toolchain guy, so I have no idea what that library is about.  @jeremybennett, can you shed any light on this?

Try this:
```
opt/riscv/bin/riscv32-unknown-elf-gcc -v
```
You should get something very much like this:
```
Using built-in specs.
COLLECT_GCC=/opt/riscv/bin/riscv32-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/opt/riscv32-embecosm-gcc-ubuntu2004-20211027/bin/../libexec/gcc/riscv32-unknown-elf/12.0.0/lto-wrapper
Target: riscv32-unknown-elf
Configured with: ../../gcc/configure --target=riscv32-unknown-elf --prefix=/build/workspace/riscv32-gcc-ubuntu2004/install --with-sysroot=/build/workspace/riscv32-gcc-ubuntu2004/install/riscv32-unknown-elf --with-native-system-header-dir=/include --with-newlib --disable-shared --enable-languages=c,c++ --enable-tls --disable-werror --disable-libmudflap --disable-libssp --disable-quadmath --disable-libgomp --disable-nls --enable-multilib --with-multilib-generator='rv32e-ilp32e--c rv32ea-ilp32e--m rv32em-ilp32e--c rv32eac-ilp32e-- rv32emac-ilp32e-- rv32i-ilp32--c rv32ia-ilp32--m rv32im-ilp32--c rv32if-ilp32f-rv32ifd-c rv32iaf-ilp32f-rv32imaf,rv32iafc-d rv32imf-ilp32f-rv32imfd-c rv32iac-ilp32-- rv32imac-ilp32-- rv32imafc-ilp32f-rv32imafdc- rv32ifd-ilp32d--c rv32imfd-ilp32d--c rv32iafd-ilp32d-rv32imafd,rv32iafdc- rv32imafdc-ilp32d-- rv64i-lp64--c rv64ia-lp64--m rv64im-lp64--c rv64if-lp64f-rv64ifd-c rv64iaf-lp64f-rv64imaf,rv64iafc-d rv64imf-lp64f-rv64imfd-c rv64iac-lp64-- rv64imac-lp64-- rv64imafc-lp64f-rv64imafdc- rv64ifd-lp64d--m,c rv64iafd-lp64d-rv64imafd,rv64iafdc- rv64imafdc-lp64d--' --with-arch=rv32imac --with-abi=ilp32 --with-bugurl=''\''https://www.embecosm.com'\''' --with-pkgversion=''\''riscv32-embecosm-gcc-ubuntu2004-20211027'\'''
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 12.0.0 20211027 (experimental) ('riscv32-embecosm-gcc-ubuntu2004-20211027') 
```
The above indicates that I am running with `riscv32-embecosm-gcc-ubuntu2004-20211027`.

mohamedElbouazzati, 2021-12-17:
Thanks for your answer.
Here is the version of the TC:

` /opt/riscv/bin/riscv32-unknown-elf-gcc -v
`
```
Using built-in specs.
COLLECT_GCC=/opt/riscv/bin/riscv32-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/opt/riscv/bin/../libexec/gcc/riscv32-unknown-elf/12.0.0/lto-wrapper
Target: riscv32-unknown-elf
Configured with: ../../gcc/configure --target=riscv32-unknown-elf --prefix=/build/workspace/riscv32-gcc-ubuntu2004/install --with-sysroot=/build/workspace/riscv32-gcc-ubuntu2004/install/riscv32-unknown-elf --with-native-system-header-dir=/include --with-newlib --disable-shared --enable-languages=c,c++ --enable-tls --disable-werror --disable-libmudflap --disable-libssp --disable-quadmath --disable-libgomp --disable-nls --enable-multilib --with-multilib-generator='rv32e-ilp32e--c rv32ea-ilp32e--m rv32em-ilp32e--c rv32eac-ilp32e-- rv32emac-ilp32e-- rv32i-ilp32--c rv32ia-ilp32--m rv32im-ilp32--c rv32if-ilp32f-rv32ifd-c rv32iaf-ilp32f-rv32imaf,rv32iafc-d rv32imf-ilp32f-rv32imfd-c rv32iac-ilp32-- rv32imac-ilp32-- rv32imafc-ilp32f-rv32imafdc- rv32ifd-ilp32d--c rv32imfd-ilp32d--c rv32iafd-ilp32d-rv32imafd,rv32iafdc- rv32imafdc-ilp32d-- rv64i-lp64--c rv64ia-lp64--m rv64im-lp64--c rv64if-lp64f-rv64ifd-c rv64iaf-lp64f-rv64imaf,rv64iafc-d rv64imf-lp64f-rv64imfd-c rv64iac-lp64-- rv64imac-lp64-- rv64imafc-lp64f-rv64imafdc- rv64ifd-lp64d--m,c rv64iafd-lp64d-rv64imafd,rv64iafdc- rv64imafdc-lp64d--' --with-arch=rv32imac --with-abi=ilp32 --with-bugurl=''\''https://www.embecosm.com'\''' --with-pkgversion=''\''riscv32-embecosm-gcc-ubuntu2004-20211212'\'''
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 12.0.0 20211211 (experimental) ('riscv32-embecosm-gcc-ubuntu2004-20211212') 
```

MikeOpenHWGroup, 2021-12-19:
I see that you are using a more up-to-date version of the toolchain (20211212) than I had (20211027), so I thought perhaps that could be the issue.  However updating my toolchain to 20211212 does not change anything for me.   (This is actually _good_ news because it means the toolchain is not the problem.)

At this point I am all out of ideas, so I will consult with some colleagues on Monday.

mohamedElbouazzati, 2021-12-19:
OK, thanks, I'll check on my side to see if I missed something.

mohamedElbouazzati, 2021-12-20:
Hello, I found that the BSP directory for cv32e40p has already been compiled by me, and I did a first cleanup and went to the /sim/core directory so the helloworld test works fine for me, it was a mistake on my part I compiled the BSP before... just to try my toolchain... 

So sorry and thanks.

[image]

closed 2021-12-20 by mohamedElbouazzati

MikeOpenHWGroup, 2021-12-20:
Thanks @mohamedElbouazzati.  I should have thought to ask about this.

For future reference, there is a make target `clean_all` which will remove _all_ generated files plus the cloned RTL.   It can be very helpful.