openhwfoundation/core-v-verif #1082: Default CV_SW_MARCH not supporting 40x rand tests

silabs-robin, 2021-12-21:
### Type
* Compile error

### Steps to Reproduce
1. On latest `cv32e40x/dev` branch (399c44 as of now).
2. Set environment variables `CV_SIMULATOR`, `CV_SW_TOOLCHAIN`, `CV_CORE` etc (but not CV_SW_MARCH).
3. Run `./bin/makeuvmt comp_corev-dv gen_corev-dv test TEST=corev_rand_fencei CV_CORE=cv32e40x`.
4. Problem, `Error: unrecognized opcode` is occuring for several B-ext instructions.

### Additional context
A workaround is setting `CV_SW_MARCH` env variable to `rv32imc_zba1p00_zbb1p00_zbc1p00_zbs1p00`.
Note, the regressions seemingly don't have this problem.
I don't think we want to make `CV_SW_MARCH` a non-optional env var to set?
In `core-v-verif/cv32e40x/tests/cfg/default.yaml` the correct march is specified, but it seems like it is ineffectful when running sims from cli.

I tried looking at the makefile to see what goes wrong but it was too complicated to figure out atm so I created this issue instead.

As a side note, we might want to decide about the make env variables, make it clear which are public ("CV_" ones) vs private, and also which are required vs optional.
The [mk/uvmt readme](https://github.com/openhwgroup/core-v-verif/blob/master/mk/uvmt/README.md) says CV_SW_MARCH defaults to "unknown" but that is both untrue and not a legal value, and some additional vars should maybe be marked "optional".

Relates to changes made in PR https://github.com/openhwgroup/core-v-verif/pull/1077.
Relates to discussion in issue https://github.com/openhwgroup/core-v-verif/issues/1052.

MikeOpenHWGroup, 2021-12-21:
Thanks for bringing this up up @silabs-robin.

> I don't think we want to make CV_SW_MARCH a non-optional env var to set?

Agreed.  It should be an option we provide (just in case someone wants to hang themselves), but it should not be required.

> In core-v-verif/cv32e40x/tests/cfg/default.yaml the correct march is specified, but it seems like it is ineffectful when running sims from cli.

I believe this is the crux of the problem.  I should have some time to look at this today.

> I tried looking at the makefile to see what goes wrong but it was too complicated to figure out atm so I created this issue instead.

Welcome to my nightmare.  In general, I am a fan of Makefiles, but we are pushing them to an illogical extreme.  Some well placed "helper scripts" written in python would be helpful.

> The mk/uvmt readme says CV_SW_MARCH defaults to "unknown" but that is both untrue and not a legal value, and some additional vars should maybe be marked "optional".

Hmmm.  Needs to be fixed.  I will also look at that.

MikeOpenHWGroup, 2021-12-22:
Update.  I believe the root of the problem reported in this issue is on [line 294 of mk/Common.mk](https://github.com/openhwgroup/core-v-verif/blob/master/mk/Common.mk#L294).  That line reads:

```
RISCV_MARCH       = $(call RESOLVE_FLAG2,$(TEST_RISCV_MARCH),$(CV_SW_MARCH))
```
In order to override it with the RISCV_MARCH in the yaml configuration it should be:

```
RISCV_MARCH       = $(call RESOLVE_FLAG2,$(CFG_RISCV_MARCH),$(CV_SW_MARCH))
```

I have tried this on my desktop machine (running Metrics dsim) and on a CMC VM (running Cadence Xcelium) and it works for both.  Let me know if that works for you and I'll generate a pull-request.

The reason I haven't already issued a pull-request is that methodology for selecting toolchains is in serious need of refactoring.  I have some ideas on this, but they are not yet ready to be published, so I'd like to "hot fix" the immediate problem and generate a longer term solution over the holidays.

silabs-hfegran, 2021-12-27:
> Update. I believe the root of the problem reported in this issue is on [line 294 of mk/Common.mk](https://github.com/openhwgroup/core-v-verif/blob/master/mk/Common.mk#L294). That line reads:
>
> ``` [...]

Hi @MikeOpenHWGroup, line 279:
```
CV_SW_MARCH ?= $(call RESOLVE_FLAG2,$(CFG_CV_SW_MARCH),rv32imc)
```
Like all the other configs, sets the march to rv32imc if it is not overridden in the configuration. 
Line 294, which you mentioned, overrides this parameter with a TEST-specific one. 

Thus, to fix this issue, what you want to do is to add the following to the yaml-files:
```
cv_sw_march: rv32imc_zba1p00_zbb1p00_zbc1p00_zbs1p00
```

I this way, one also avoids having to look into the makefile to see which architecture spec is in use.
For some reason we have 5 different setups, GNU, COREV, PULP, LLVM and CV_SW.

MikeOpenHWGroup, 2021-12-28:
You are right @silabs-hfegran, this can be fixed without modifying the `mk/Common.mk` Makefile.  Thanks for the clarification.  The question now is, where should the configuration be?  There are (at least) two options:
1. `cv32e40x/tests/cfg/default.yaml`.  This will set cv_sw_march for all test-programs that use the default test-program configuration.
2. `cv32e40x/tests/programs/corev-dv/corev_rand_fencei/test.yaml`.  This will set cv_sw_march for the corev_rand_fencei test-program only.

Either will work.  The first option will cause the toolchain to generate/compile bitmanip instructions whenever possible.  This may or may not be what you want.  You are the verification lead on the E40X so I will defer to your decision on this.

> For some reason we have 5 different setups, GNU, COREV, PULP, LLVM and CV_SW.

An historical artifact. As I stated above, the methodology for selecting toolchains is in serious need of refactoring.  I have a solution to propose but it must wait until we resolve this issue.  (I _think_ we are there.)

silabs-hfegran, 2022-01-03:
> You are right @silabs-hfegran, this can be fixed without modifying the `mk/Common.mk` Makefile. Thanks for the clarification. The question now is, where should the configuration be? There are (at [...]

Good question @MikeOpenHWGroup

The path I have followed with my update to the configurations & makefiles is that the largest supported set of extensions is the default (configuration-wide) setting, while certain extensions can be disabled on a per-test basis (or if deemed necessary, its own configuration). Thus, my proposition is that the cfg/*.yaml-files dictate the standard compiler architecture settings, while specific tests are free to override this if needed.

This is also why I created the specific configuration that specifically disables the bitmanip extensions (no_bitmanip).
PR #1077 should have resolved this issue, but the specific setting for CV_SW-toolchain apparently got omitted.

silabs-hfegran, 2022-01-03:
Alternatively, one could hard-code the default inside the makefiles (The makefile as you noticed, uses rv32imc by default), though I prefer the increased visibility with the default setting directly inside the configuration yaml, as this will make it apparent at first glance which compiler setting is used. 

In my experience, the more "standard" settings that are already in the configuration files (instead of depending on hard coded makefile variables), the easier it is for someone new to pick this up.

MikeOpenHWGroup, 2022-01-04:
PR #1092 is a proposed refactoring of the common makefile to both simplify and resolve this issue.

closed 2022-02-03 by silabs-hfegran