openhwfoundation/core-v-verif #1110: RVVI and RVFI agent incompatibility across CV32E40X/E40S

MikeOpenHWGroup, 2022-01-14:
### Type
Compile error of the CV32E40S in the cv32e40x/dev branch.

As we were discussing this morning, I wanted to look into whether all cores would run in a specific core's "dev" branch.   In this case, the cv32e40x/dev branch.  If you try compiling the sanity test of the CV32E40X on the `cv32e40x/dev` branch of `git@github.com:openhwgroup/core-v-verif.git`, it will compile and run correctly, but the e40s will not compile at all.   In an attempt to rectify this, I found what seems to be an incompatibility of RVVI and RVFI across both the E40X and E40S.

### Steps to Reproduce
On my fork of core-v-verif there is a branch `cv32e40x_dev_incompatible_with_e40s` that illustrates the point.  This branch updates to the RVVI and RVFI agents to be compatible with the E40S environment.
```
$ git clone -b cv32e40x_dev_incompatible_with_e40s git@github.com:MikeOpenHWGroup/core-v-verif.git tryit
$ cd tryit/cv32e40s/sim/uvmt
$ make sanity SIMULATION=<your_fav_simulator>
```
The above will work.  But if you try to run the e40x, you will get compile-time failures:
```
$ cd ../../../cv32e40x/sim/uvmt
$ make sanity SIMULATION=<your_fav_simulator>
```
If you try the above on the `cv32e40x/dev` branch of `git@github.com:openhwgroup/core-v-verif.git`, you will get the opposite results: the 40x will compile and run correctly, but the e40s will not compile.

### Additional context
Affected files are:
```
lib/uvm_agents/uvma_rvfi/seq/uvma_rvfi_instr_seq_item.sv
lib/uvm_agents/uvma_rvfi/uvma_rvfi_cfg.sv
lib/uvm_agents/uvma_rvfi/uvma_rvfi_instr_mon.sv
lib/uvm_agents/uvma_rvfi/uvma_rvfi_mon_trn_logger.sv
lib/uvm_agents/uvma_rvvi/uvma_rvvi_cfg.sv
lib/uvm_agents/uvma_rvvi_ovpsim/seq/uvma_rvvi_ovpsim_control_seq.sv
lib/uvm_agents/uvma_rvvi_ovpsim/seq/uvma_rvvi_ovpsim_control_seq_item.sv
lib/uvm_agents/uvma_rvvi_ovpsim/uvma_rvvi_ovpsim_drv.sv
```

silabs-robin, 2022-01-26:
> On my fork of core-v-verif there is a branch cv32e40x_dev_incompatible_with_e40s that illustrates the point. This branch updates to the RVVI and RVFI agents to be compatible with the E40S environment.

One question about this.
Does the branch make any changes to those `uvma_rvvi` files mentioned above?
When I look at the latest commit and on the commit tree, I only see commits from the 40x dev branch itself, and no new temporary-changes commits.

silabs-hfegran, 2024-01-11:
@MikeOpenHWGroup Please close this task if this is resolved - I believe that it should be though the occasional breakage may occur

MikeOpenHWGroup, 2024-01-11:
Ha!  By coincidence, I was checking on this today.  It appears that some of that "occasional breakage" has occurred, so I will investigate before closing this Issue.