openhwfoundation/core-v-verif #1122: Unexpected DSIM error: "No clocking event is specified or can be inferred"

MikeOpenHWGroup, 2022-01-17:
A recent update to `uvmt_cv32e40x_interrupt_assert.sv`, see [line 205](https://github.com/openhwgroup/core-v-verif/blob/a0b44fd155dad5bd03a1039e88a0b611e21fa2be/cv32e40x/tb/uvmt/uvmt_cv32e40x_interrupt_assert.sv#L205) and [line 432](https://github.com/openhwgroup/core-v-verif/blob/a0b44fd155dad5bd03a1039e88a0b611e21fa2be/cv32e40x/tb/uvmt/uvmt_cv32e40x_interrupt_assert.sv#L430), causes compile errors under dsim:
```
=E:[MissingClockingEvent]:
    No clocking event is specified or can be inferred    

    /home/mike/GitHubRepos/strichmo/core-v-verif/cv32e40x/merge_dev_to_rel_57/cv32e40x/tb/uvmt/uvmt_cv32e40x_interrupt_assert.sv:207:21

[snip]
```
The code in question compiles and simulates under Xcelium.  I do not see a reason for this error, so a hot-fix will be applied to exclude these code segments from dsim compilation.  As these are coverage properties, the fidelity of the testing will not be affected (but coverage will).   An issue will be raised with Metrics as well.

MikeOpenHWGroup, 2022-01-17:
This has been reported to Metrics and can be referenced as `request #1163`.

silabs-robin, 2022-01-18:
Hi @MikeOpenHWGroup. That error message seems a bit strange.
Looking at the git blame of that file, isn't that line 10 months old?
And this file has a "default clocking", so how could there be no clocking event?
I don't really understand that, but maybe the people at Metrics are able to spot the problem.

MikeOpenHWGroup, 2022-02-03:
This issue can be closed as soon as #1161 is merged in.

PR #1161 (2022-02-03): New version of DSIM resolves #1122
This PR brings in an updated version of dsim that resolves #1122 (and removes the hack exclude the TB code that the previous dsim release didn't like).

For the record, the Metrics issue number for this is 1163 (that is not a GitHub issue number!).

This PR has a DO NOT MERGE lable on it as I just noticed a mistake with the .metrics.json that will impact the cv32e40s.  I will fix it before removing the label.
commit bda04485 (2022-02-03): Merge pull request #1161 from MikeOpenHWGroup/cv32e40x/dev

New version of DSIM resolves #1122
commit ec0eaf4d (2022-02-03): dsim 20211122.7.0-02Feb2022 resolves #1122
closed 2022-02-03 by silabs-hfegran