openhwfoundation/core-v-verif #1123: Merge conflicts: "40p_release" -> "master"

silabs-robin, 2022-01-18:
As part of the merge procedure I merged "40p_release" into "master" (on my own fork), but there was merge conflicts.
@MikeOpenHWGroup could you have a look to see if I did my manual conflict resolution correctly?
Here is my branch https://github.com/openhwgroup/core-v-verif/compare/master...silabs-robin:silabs-robin/pr/master_merge_220118_40p
And the latest commit (the merge commit) is the one of interest https://github.com/openhwgroup/core-v-verif/commit/09b938b059681e332700d4516741b404743fa426.

--------

Here is a summary of the files I had to resolve conflicts for:

Problem: "docs/CodingStyleGuidelines.md" modified in "master" and "40p".
Which is correct?
I chose the latest of those two edits.
(Shall/Should/May/Can part.)

Problem: "README.md" modified in "master" and "40p".
Which is correct?
I did a best guess.
("## Getting Started" part.)

Problem: "cv32e40p/sim/README.md" modified in "master" and "40p".
Which is correct?
I chose the latest one.
("Directories" part.)

Problem: "cv32e40p/sim/core/.gitignore" modified in "master" and "40p".
Which is correct?
I chose the latest one.
("*_results" part.)

Problem: "cv32e40p/sim/core/Makefile" modified in "master" and "40p".
Which is correct?
I did a best guess, trying to not lose anything, trying to use latest commits.
(Several parts changed.)

silabs-robin, 2022-01-18:
I am continuing the `silabs-robin/pr/master_merge_220118_40p` merging in `silabs-robin/pr/master_merge_220118_40p_40x`, where I shall pull in from `cv32e40x/release`.
So if you do see something you want to change, I don't think it matters whether you PR to my `..._40p` or `..._40p_40x` branch because I can plan to continue on `..._40p_40x_40s` and I can just pull in changes from whichever of the previous ones AFAICT.

MikeOpenHWGroup, 2022-01-18:
Hello @silabs-robin, good work on this.

> Problem: "docs/CodingStyleGuidelines.md" modified in "master" and "40p".
> Which is correct?
> I chose the latest of those two edits.
> (Shall/Should/May/Can part.)

Yes, that is correct.

> Problem: "README.md" modified in "master" and "40p".
> Which is correct?
> I did a best guess.
> ("## Getting Started" part.)

I would call your "best guess" an "educated guess" as you chose the proper ones.

> Problem: "cv32e40p/sim/README.md" modified in "master" and "40p".
> Which is correct?
> I chose the latest one.
> ("Directories" part.)

Again, you chose correctly.

> Problem: "cv32e40p/sim/core/.gitignore" modified in "master" and "40p".
> Which is correct?
> I chose the latest one.
> ("*_results" part.)

Again, you chose correctly.

> Problem: "cv32e40p/sim/core/Makefile" modified in "master" and "40p".
> Which is correct?
> I did a best guess, trying to not lose anything, trying to use latest commits.
> (Several parts changed.)

Oh, _so_ close!   Everthing is bang on except for these lines:
[image]

silabs-robin, 2022-01-19:
Darn :) Almost got it right.

In `cv32e40p/sim/core/Makefile`, should it be those red lines with `SIM_TEST_RESULTS` and `VERI_CUSTOM` that should have been used?
(If yes then I think I might have removed VERI_CUSTOM if it was used nowere else, so that has to be fixed.)

I'll leave this issue open until I find the best time to fix it.
Maybe when I have a "master" branch ready to PR, I can add a commit to get this in place first...

Nicolas-Gaudin, 2022-02-01:
I had an issue when using the veri-test command. Indeed, this is related to what you are talking about.

https://github.com/openhwgroup/core-v-verif/blob/f741f3721eddee862b8de78836fc6d80d3404f0a/cv32e40p/sim/core/Makefile#L505
	mv $(VERI_OBJ_DIR)/Vtb_top_verilator **$(SIM_TEST_RESULTS)/verilator_executable**

https://github.com/openhwgroup/core-v-verif/blob/f741f3721eddee862b8de78836fc6d80d3404f0a/cv32e40p/sim/core/Makefile#L512
	./**testbench_verilator** $(VERI_FLAGS) \

Both bold text have to be identical. After modifying it, veri-test works.

However, I don't know if I should put l.505 element in l.512 or the opposite to don't create other problems on other cases.

MikeOpenHWGroup, 2022-02-02:
Merges to/from `master` and `cv32e40p/release` branches resolved and all associated PRs are "in".

Need a a couple of hotfixes: #1157 for `cv32e40p/release` and #1158 for `master`.

@silabs-robin, I believe this issue can be closed now, but you opened it, so that is your call.

@Nicolas-Gaudin, thanks for flagging the issue with the CV32E40P Core testbench.  Sadly, this testbench receive little attention these days and it is (deliberately) not part of our CI flow.  Please do raise an issue if you encounter any problems with it.

silabs-robin, 2022-02-02:
Thanks @MikeOpenHWGroup. I will close the issue, but I want to look at a few tests first.

closed 2022-02-03 by silabs-robin