openhwfoundation/core-v-verif #1124: Merge conflicts: "40x_release" -> "master"+"40p_release"

silabs-robin, 2022-01-18:
First I merged "40p_release" into "master" (see https://github.com/openhwgroup/core-v-verif/issues/1123) and now I merged "40x_release" into that previous merge of "master"+"40p_release".
There was conflicts.

Here is the diff https://github.com/openhwgroup/core-v-verif/commit/65706bc94788b9efcb41bf20b97abe28d7e7d069 (sadly includes more lines than the ones who conflicted).
@MikeOpenHWGroup  could you see if you agree with the manual conflict resolution i did?

--------

Here is a summary of the files I had to resolve conflicts for:

Problem: ".metrics.json"
"SIMULATOR=dsim" seem to have been removed?
Picked the latest commit (removing "SIMULATOR=dsim").

Problem: "cv32e40x/README.md"
"cv32e40x" vs "CV32E40X" capitalization.
Picked uppercase because that's what P and S has.

Problem: "cv32e40x/sim/README.md"
Markdown, new sentence on new line.
Picked the nicest one.

Problem: "mk/uvmt/README.md"
Huge addition/removal.
Picked the short one, because "mk/README.md" seems to have the same info but more up to date.

MikeOpenHWGroup, 2022-01-18:
This time, you are 4 for 4.

> Problem: ".metrics.json"
> "SIMULATOR=dsim" seem to have been removed?
> Picked the latest commit (removing "SIMULATOR=dsim").

Yes, having the `SIMULATOR=dsim` argument for each command has been made redundant by the (new) [variables](https://github.com/openhwgroup/core-v-verif/blob/ee67da1f0fda04cbb59e784f0d944a1071f7058b/.metrics.json#L2) entry.

> Problem: "cv32e40x/README.md"
> "cv32e40x" vs "CV32E40X" capitalization.
> Picked uppercase because that's what P and S has.

Thank you.  That is my preference, but of course either will work in a README.

> Problem: "cv32e40x/sim/README.md"
> Markdown, new sentence on new line.
> Picked the nicest one.

Not sure what "nicest one" means, but you selected the line which references `ExternalRepos.mk`, which is the right one.  (We've had _two_ Makefiles called `Common.mk` for many months and it has been driving me crazy.)

> Problem: "mk/uvmt/README.md"
> Huge addition/removal.
> Picked the short one, because "mk/README.md" seems to have the same info but more up to date.

Yes, that was the right decision.  As you've guessed, the purpose was to move information from `mk/uvmt` to `mk` because a lot of what is there is relevant for both the core and UVM environments.  While I was moving text around I took the opportunity to update it.

silabs-robin, 2022-01-19:
Thanks for reviewing the manual edits.
I'll close this issue since there were no problems.

closed 2022-01-19 by silabs-robin