openhwfoundation/core-v-verif #1129: mhpmcounter29_csr_access tests pass with wrong configuration

silabs-hfegran, 2022-01-20:
## mhpmcounter29_csr_access tests pass with wrong configuration

mhpmcounter29_csr_access_test_* has consistently been running under the wrong configuration since the build environment enabled multiple configs for each test in regressions. Additionally, the test explicitly states that these test must be run with NUM_HPMCOUNTERS=29, and while this is the case, test_1 appears to consistently pass regression tests - this should not be the case.

### Problems
* Wrong configuration set - verify correct parameter set when using the correct configuration
* Test passes with incorrect parameters - This should not happen

### Steps to Reproduce
Please provide:
1. cv32e40x/dev, 55deb696e671451f4057d6be3b49c1f4f29c13d1
2. Command line: `cv32e40x_full regression` / `make test TEST=mhpmcounter29_csr_access_test_<1|2> CFG=<default|num_mhpmcounter29>`

silabs-hfegran, 2022-01-20:
PR #1128 should solve the cfg issue, however the ISS needs an update to set the mcountinhibit CSR correctly on reset, thus I have disbled scoreboard checking for that particular CSR.

silabs-robin, 2023-12-12:
> thus I have disbled scoreboard checking for that particular CSR.

I think that refers to this:
[image]

And it seems to have been fixed, because it no longer exists:
[image]

closed 2023-12-12 by silabs-robin