openhwfoundation/core-v-verif #1138: Cover the cross of NMI and Debug with assertions

silabs-mateilga, 2022-01-21:
The cases where an entry to debug crosses with an NMI event are not covered by assertions. There are several corner cases here, which should be covered.

Note that the corner case where an illegal instruction, debug entry and NMI has been excluded from the assert "a_illegal_insn_debug_req". Make sure this case is covered by the assertions added in this task.

Task is completed when assertions covering any case where an NMI and a debug request interact. 
Develop for cv32e40x.

MikeOpenHWGroup, 2022-01-21:
> The cases where an entry to debug crosses with an NMI event are not covered by assertions. 

Are the cases where an entry to debug crosses with an NMI event mentioned in the [DVplan(s)](https://github.com/openhwgroup/core-v-verif/tree/master/cv32e40x/docs/VerifPlans/Simulation)?  If not, then please create an issue against the DVplan.  If these cases _are_ covered in the DVplan, then I recommend we close this issue as it is part of our process to have either a directed testcase or (preferably) functional coverage (either coverpoints or cover properties) for all features of the DUT.

silabs-robin, 2023-12-12:
@silabs-mateilga This covers it?

[image]

closed 2023-12-12 by silabs-robin