openhwfoundation/core-v-verif #1176: Error in environment documentation, deferint signal

silabs-mateilga, 2022-02-10:
the description of the _deferint_ signal on line 850 specifies that the signal is active low, which does not match the documentation of the model provided by Imperas, that specifies the signal is active high.

Local ref:
core-v-verif/docs/VerifStrat/source/cv32_env.rst

Imperas ref(when checked out):
core-v-verif/vendor_lib/imperas/imperas_DV_COREV/doc/OVP_Model_Specific_Information_openhwgroup_riscv_CV32E40X.pdf

MikeOpenHWGroup, 2022-02-10:
Thanks for reporting this @silabs-mateilga.   I assume that this is non-blocking, please advice if it _is_ blocking you (or anyone else).

Blocking or not, its important to clear this up.  I will look into it.  Do you have a testcase on either the E40S or E40X that asserts _deferint_?

silabs-mateilga, 2022-02-10:
@MikeOpenHWGroup  This problem is in documentation only, and is therefore only confusing, not blocking. I'm not sure if it is necessary to see a testcase, but the signal is driven by the methods in uvma_rvvi_ovpsim_drv.sv to control the reference model, so any test that runs interrupts would show activity on the signal. I'm currently working on corev_rand_instr_obi_err, as an example, but any interrupt test case would do.

silabs-robin, 2023-12-11:
@MikeOpenHWGroup do you know if this can be closed?

silabs-mateilga, 2023-12-13:
Outdated issue, closing.