openhwfoundation/core-v-verif #1185: new-agent fatals out

MikeOpenHWGroup, 2022-02-14:
Running the new-agent Makefile target generates _most_ of the shell for an Agent, but fatals out over something called `agent_simplex_no_layers`.

### Steps to Reproduce
```
$ git clone https://github.com/openhwgroup/core-v-verif master
$ cd master/cv32e40p/sim/uvmt
$ make new-agent
```

### Additonal Context
Example run of `new-agent` follows:
```
/home/mike/GitHubRepos/openhwgroup/core-v-verif/master/mk/Common.mk:83: Must define a DPI_DASM_SPIKE_REPO to use the common makefile
/home/mike/GitHubRepos/openhwgroup/core-v-verif/master/mk/Common.mk:278: CV_SW_MARCH not defined in either the shell environment, test.yaml or cfg.yaml
/home/mike/GitHubRepos/openhwgroup/core-v-verif/master/mk/Common.mk:291: CV_SW_CC not defined in either the shell environment, test.yaml or cfg.yaml
/home/mike/GitHubRepos/openhwgroup/core-v-verif/master/mk/Common.mk:303: CV_SW_CFLAGS not defined in either the shell environment, test.yaml or cfg.yaml
/home/mike/GitHubRepos/openhwgroup/core-v-verif/master/mk/Common.mk:318: RISCV set to /opt/riscv
/home/mike/GitHubRepos/openhwgroup/core-v-verif/master/mk/Common.mk:319: RISCV_PREFIX set to riscv32-unknown-elf-
/home/mike/GitHubRepos/openhwgroup/core-v-verif/master/mk/Common.mk:320: RISCV_EXE_PREFIX set to /opt/riscv/bin/riscv32-unknown-elf-
/home/mike/GitHubRepos/openhwgroup/core-v-verif/master/mk/Common.mk:321: RISCV_MARCH set to rv32imc
/home/mike/GitHubRepos/openhwgroup/core-v-verif/master/mk/Common.mk:322: RISCV_CC set to gcc
/home/mike/GitHubRepos/openhwgroup/core-v-verif/master/mk/Common.mk:323: RISCV_CFLAGS set to 
mkdir -p /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/temp
wget --no-check-certificate -q https://mooreio.com/packages/uvm_gen.tgz -P /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/temp
tar xzf /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/temp/uvm_gen.tgz -C /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/temp
cd /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/temp && ./src/new_agent_simplex_no_layers.py /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/lib/uvm_agents "OpenHW Group"
Code will be output to /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/lib/uvm_agents
Default copyright owner is OpenHW Group
Moore.io Template Generator: UVM Agent - Simplex, no layers (v1p0)
******************************************************************
The answers to the following questionnaire will be used to generate the code for your new UVM Agent

Please enter a specific name for the copyright holder or hit RETURN for the default (default is 'OpenHW Group'):
abc
Please enter the package name for this agent (ex: 'apb'); this name will be used for all agent types (ex: 'uvma_apb_agent_c'):
abc
Please enter the (descriptive) name for this agent (ex: 'Advanced Peripheral Bus (APB)'):
abc
Writing /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/lib/uvm_agents/uvma_abc/bin/package.py
Writing /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/lib/uvm_agents/uvma_abc/docs/agent_block_diagram.svg
Writing /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/lib/uvm_agents/uvma_abc/examples/instantiation.sv
Writing /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/lib/uvm_agents/uvma_abc/examples/sequence.sv
Writing /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/lib/uvm_agents/uvma_abc/src/comps/uvma_abc_agent.sv
Writing /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/lib/uvm_agents/uvma_abc/src/comps/uvma_abc_cov_model.sv
Writing /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/lib/uvm_agents/uvma_abc/src/comps/uvma_abc_drv.sv
Writing /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/lib/uvm_agents/uvma_abc/src/comps/uvma_abc_mon_trn_logger.sv
Writing /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/lib/uvm_agents/uvma_abc/src/comps/uvma_abc_mon.sv
Writing /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/lib/uvm_agents/uvma_abc/src/comps/uvma_abc_seq_item_logger.sv
Writing /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/lib/uvm_agents/uvma_abc/src/comps/uvma_abc_sqr.sv
Writing /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/lib/uvm_agents/uvma_abc/src/obj/uvma_abc_cfg.sv
Writing /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/lib/uvm_agents/uvma_abc/src/obj/uvma_abc_cntxt.sv
Writing /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/lib/uvm_agents/uvma_abc/src/obj/uvma_abc_mon_trn.sv
Writing /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/lib/uvm_agents/uvma_abc/src/seq/uvma_abc_base_seq.sv
Writing /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/lib/uvm_agents/uvma_abc/src/seq/uvma_abc_seq_item.sv
Writing /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/lib/uvm_agents/uvma_abc/src/uvma_abc_constants.sv
Writing /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/lib/uvm_agents/uvma_abc/src/uvma_abc_if_chkr.sv
Writing /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/lib/uvm_agents/uvma_abc/src/uvma_abc_if.sv
Writing /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/lib/uvm_agents/uvma_abc/src/uvma_abc_macros.sv
Writing /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/lib/uvm_agents/uvma_abc/src/uvma_abc_pkg.flist
Writing /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/lib/uvm_agents/uvma_abc/src/uvma_abc_pkg.flist.xsim
Writing /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/lib/uvm_agents/uvma_abc/src/uvma_abc_pkg.sv
Writing /home/mike/GitHubRepos/openhwgroup/core-v-verif/master/lib/uvm_agents/uvma_abc/src/uvma_abc_tdefs.sv
Traceback (most recent call last):
  File "./src/new_agent_simplex_no_layers.py", line 198, in <module>
    process_all_files()
  File "./src/new_agent_simplex_no_layers.py", line 125, in process_all_files
    process_file(file, files[file])
  File "./src/new_agent_simplex_no_layers.py", line 99, in process_file
    fin = open(relative_path_to_template + path, "rt")
FileNotFoundError: [Errno 2] No such file or directory: '/home/mike/GitHubRepos/openhwgroup/core-v-verif/master/temp/templates/agent_simplex_no_layers/.gitignore'
make: *** [/home/mike/GitHubRepos/openhwgroup/core-v-verif/master/mk/Common.mk:406: new-agent] Error 1
$ 
```

datum-dpoulin, 2022-02-23:
Fixed committed to mio repo and tested.

MikeOpenHWGroup, 2022-02-23:
Confirmed.   Thanks!