openhwfoundation/core-v-verif #1193: OVPSIM mret does not set MIE = MPIE in MSTATUS CSR

silabs-mateilga, 2022-02-18:
For a few select seeds when running the "corev_rand_interrupt_wfi_mem_stress" test, what seems to be a bug in the reference model can be observed. The scenario is that when a core is woken from a wfi instruction by a disabled interrupt, and the next instruction is an mret instruction, the MIE bit is not set to MPIE in the MSTATUS CSR, which is in breach with the 1.12 privileged spec.

Steps to Reproduce:
Branch: cv32e40x/dev  https://github.com/openhwgroup/core-v-verif/tree/cv32e40x/dev
Hash: fe9bdeff9178adbff54ca5591e71e08c757bffae

Run commands:
make comp_corev-dv gen_corev-dv TEST=corev_rand_interrupt_wfi_mem_stress CFG=default SIMULATOR=xrun USE_ISS=YES COV=YES RUN_INDEX=27 GEN_START_INDEX=27 RNDSEED=-844432648

make test TEST=corev_rand_interrupt_wfi_mem_stress CFG=default SIMULATOR=xrun USE_ISS=YES COV=YES RUN_INDEX=27 GEN_START_INDEX=27 RNDSEED=-844432648

Waveform showing the difference in the MSTATUS register between RVFI and RVVI:
[image]

@eroom1966 Could you take a look at this?

eroom1966, 2022-02-18:
@silabs-mateilga 
Having trouble reproducing your steps, this is what I did after setting my environment to run xcelium

$ git clone https://github.com/openhwgroup/core-v-verif
$ cd core-v-verif
$ git checkout cv32e40x/dev
$ git reset fe9bdef --hard
$ make comp_corev-dv gen_corev-dv TEST=corev_rand_interrupt_wfi_mem_stress CFG=default SIMULATOR=xrun USE_ISS=YES COV=YES RUN_INDEX=27 GEN_START_INDEX=27 RNDSEED=-844432648

make: *** No rule to make target 'comp_corev-dv'. Stop.

Please advise
Thx
Lee

silabs-mateilga, 2022-02-18:
Ah, apologies, I sometimes forget how much of the environment is "just there". 

export CV_SW_TOOLCHAIN=/tool/gcc/riscv32-embecosm-gcc-centos7-20211031
export CV_SW_PREFIX=riscv32-unknown-elf-
export CV_SW_VENDOR=unknown
export CV_SIMULATOR=xrun
export CV_CORE=cv32e40x

eroom1966, 2022-02-18:
next issue ....

eroom1966, 2022-02-18:
Next issue
I am sure I have reported this before, I think the problem is the type of shell

There is a 'for loop' construct used here which has a pre-requisite on the type of shell being used to invoke make
I think this is assuming csh or maybe ksh ?
I am using /bin/bash

If there are constructs requiring specific shell features, then the Makefiles must define the SHELL variable in order to use the correct interpreter
https://www.gnu.org/software/make/manual/html_node/Choosing-the-Shell.html

Do you know what shell this is expecting to invoke ?

( We could probably benefit with some advice from @MikeOpenHWGroup )

```
make: [/home/moore/git/bugs/core-v-verif/mk/uvmt/xrun.mk:446: gen_corev-dv] Error 1 (ignored)
for (( idx=27; idx < $((27 + 1)); idx++ )); do \
	cp -f /home/moore/git/bugs/core-v-verif/cv32e40x/bsp/link_corev-dv.ld /home/moore/git/bugs/core-v-verif/cv32e40x/sim/uvmt/xrun_results/default/corev_rand_interrupt_wfi_mem_stress/$idx/test_program/link.ld; \
	cp /home/moore/git/bugs/core-v-verif/cv32e40x/sim/uvmt/xrun_results/default/corev-dv/corev_rand_interrupt_wfi_mem_stress/corev_rand_interrupt_wfi_mem_stress_$idx.S /home/moore/git/bugs/core-v-verif/cv32e40x/sim/uvmt/xrun_results/default/corev_rand_interrupt_wfi_mem_stress/$idx/test_program; \
done
```

silabs-hfegran, 2022-02-18:
I believe that is plain sh-syntax

silabs-mateilga, 2022-02-18:
I'm unsure of what the specific problem is, but here is the latest documentation of the environment variables required.

https://github.com/openhwgroup/core-v-verif/tree/master/mk#required-corev-environment-variables

silabs-hfegran, 2022-02-18:
This should run fine invoked from a bash shell on most linux-distributions. 

What kind of OS are you attempting to run this on @eroom1966?

eroom1966, 2022-02-18:
I see the problem, I have constructed a small testcase
This syntax does NOT work in a bourne (/bin/sh) shell, but does work in a bourne-again shell (/bin/bash)

what you may have on your Linux system is a symbolic link from 
/bin/sh -> /bin/bash 
could you check ?

make by default will use /bin/sh - this syntax will not work in /bin/sh
in the makefile I can force it to use /bin/bash by defining
SHELL = /bin/bash

here is my makefile

```
SHELL = /bin/bash
GEN_START_INDEX=0
GEN_NUM_TESTS=10
all:
	echo "shell=$(SHELL)"
	for (( idx=${GEN_START_INDEX}; idx < $$((${GEN_START_INDEX} + ${GEN_NUM_TESTS})); idx++ )); do \
		echo "$$idx"; \
	done
```

I am pretty sure that your linux system has a link from /bin/sh -> /bin/bash
hence it gets a bash style interpreter in Make 

This is not guaranteed to work on a machine without this link if this is the case

silabs-hfegran, 2022-02-18:
Thanks for that @eroom1966, you are correct - it is linked to bash

eroom1966, 2022-02-18:
OK, the makefiles need to define the SHELL variable for the interpreter intended to be used, which it appears is /bin/bash

silabs-hfegran, 2022-02-18:
Apologies for these issues - Ubuntu to my knowledge has not been extensively tested as it is not an officially supported cadence platform. That said, this should not be related to the shell issues you are seeing. ExternalRepos.mk (which is included in  core-v-verif/cv32e40x/sim/uvmt/Makefile that should be called when executing ) exports the SHELL-variable to /bin/bash when invoked. 

Just to double check, are you running from the core-v-verif/cv32e40x/sim/uvmt-folder as @mateilga mentioned previously in this discussion? I can see this problem arising if the makefiles in mk/uvmt are invoked directly.

eroom1966, 2022-02-18:
slow progress - now I get the following message when trying to compile

eroom1966, 2022-02-18:
slow progress, getting a compilation error 
```
/home/moore/riscv/corev-openhw-gcc-ubuntu1804-20200913/bin/riscv32-corev-elf-gcc -Os -g -static -mabi=ilp32 -march=rv32imc_zba1p00_zbb1p00_zbc1p00_zbs1p00 -Wall -pedantic  -c /home/moore/git/bugs/core-v-verif/cv32e40x/bsp/crt0.S -o crt0.o
Assembler messages:
Fatal error: -march=rv32imc_zba1p0_zbb1p0_zbc1p0_zbs1p0: Invalid or unknown z ISA extension: 'zba'
```

I am guessing my toolchain is too old ?
Can you please provide a link to the toolchain necessary to compile this example ?

Thx
Lee

eroom1966, 2022-02-18:
I can now build with this toolchain - thanks

eroom1966, 2022-02-18:
@silabs-mateilga 
Approximately how many instructions before there is a difference ?

eroom1966, 2022-02-18:
Hmm, I ran this and it passed ?
My logfile attached, any indications of what I am doing wrong in not reproducing a FAILED run ?
[logfile.txt](https://github.com/openhwgroup/core-v-verif/files/8099254/logfile.txt)

silabs-mateilga, 2022-02-18:
The difference happens at the retirement of instruction 38861. I unfortunately don't have the opportunity to dig further in to this now, I'll have a look first thing Monday morning. What I can say is that this failure showed up on 2 of 100 runs of this test, so any difference that changes randomness of the build and/or instruction stream will likely result in a passed test.

eroom1966, 2022-02-18:
OK, lets look Monday
When you say difference, do you mean that the testbench reports a difference, and thus a FAILED test ?
or are you eyeballing a difference from the screenshot provided ?

I did not see a log at the beginning of this thread highlighting the reported difference
Thx
Lee

silabs-hfegran, 2022-02-21:
@silabs-mateilga I suggest you send Lee your generated & compiled .elf file together with instructions on how to execute the test on that elf file to rule out any possible toolchain-related differences here.

silabs-mateilga, 2022-02-21:
@eroom1966 
The following is the output from the test:
`UVM_ERROR @ 2599926.300 ns : uvme_cv32e40x_core_sb.sv(381) uvm_test_top.env.core_sb [CORESB] CSR Mismatch, order: 38861, pc: 0x00020c8e, csr: mstatus, rvfi = 0x00001888, rvvi = 0x00001880, mask = 0xffffffff
UVM_ERROR @ 2599956.300 ns : uvme_cv32e40x_core_sb.sv(305) uvm_test_top.env.core_sb [CORESB] PC Mismatch, rvfi_order: 38862, rvvi_order: 38862, rvfi.pc = 0x00004406, rvvi.pc = 0x00020c8e
UVM_ERROR @ 2599956.300 ns : uvme_cv32e40x_core_sb.sv(312) uvm_test_top.env.core_sb [CORESB] INSN Mismatch, order: 38862, rvfi.pc = 0x00004406, rvfi.insn = 0x024ea933, rvvi.insn = 0x30200073
UVM_ERROR @ 2599956.300 ns : uvme_cv32e40x_core_sb.sv(342) uvm_test_top.env.core_sb [CORESB] GPR Mismatch, order: 38862, pc: 0x00004406, rvfi_x[18] = 0x00000000, rvvi_x[18] = 0x0000003d
UVM_ERROR @ 2599956.300 ns : uvme_cv32e40x_core_sb.sv(381) uvm_test_top.env.core_sb [CORESB] CSR Mismatch, order: 38862, pc: 0x00004406, csr: mstatus, rvfi = 0x00001888, rvvi = 0x00001880, mask = 0xffffffff
`
As you can see, the error is reported as a failure on it's own, and the difference in mstatus causes a mismatch on the following instruction retirement as well.

I'll put together an .elf file and instructions as @silabs-hfegran suggested and send it to you.

silabs-mateilga, 2022-02-21:
[test_program.zip](https://github.com/openhwgroup/core-v-verif/files/8108359/test_program.zip)

@eroom1966 Extract these files to [your_results_path]/results/xrun_results/default/corev_rand_interrupt_wfi_mem_stress/27/test_program/ 
Run the second command in the top post here. (make test ...)

This will allow you to at least run the same instruction stream, it does however still not guarantee that you will see the same results, as the timing of the interrupts are influenced by simulator version. We run xcelium version 21.07.a001.

I think the easiest way for you to reproduce the error is if you are able to run your own regressions. What I did to find the original seed was to run an edited version of the "interrupt" regression list, where I only ran 100 runs of the failing test. This should produce at least one failure. I've seen quite consistently 1/50 test runs fail.

eroom1966, 2022-02-21:
@silabs-mateilga

eroom1966, 2022-02-21:
@silabs-mateilga 
I have unpacked your example to update the following files 

>   inflating: test_program/corev_rand_interrupt_wfi_mem_stress_27.elf  
>   inflating: test_program/corev_rand_interrupt_wfi_mem_stress_27.hex  
>   inflating: test_program/corev_rand_interrupt_wfi_mem_stress_27.itb  
>   inflating: test_program/corev_rand_interrupt_wfi_mem_stress_27.objdump  
>   inflating: test_program/corev_rand_interrupt_wfi_mem_stress_27.readelf  

I then ran the following command, as instructed

> make test TEST=corev_rand_interrupt_wfi_mem_stress CFG=default SIMULATOR=xrun USE_ISS=YES COV=YES RUN_INDEX=27 GEN_START_INDEX=27 RNDSEED=-844432648

The test reported PASSED, and I had no errors, logfile attached
[logfile.txt](https://github.com/openhwgroup/core-v-verif/files/8108905/logfile.txt)

I think we need to ask a third party to verify your findings, lets also compare simulator versions.
looks like I am running the following

> xrun(64): 20.03-s010: (c) Copyright 1995-2020 Cadence Design Systems, Inc.

eroom1966, 2022-02-21:
I also have 2.09 installed - trying to see if I can reproduce with that version

silabs-mateilga, 2022-02-21:
I highly doubt we will see a failure with a seed that fails on one xcelium version on another xcelium version. As I mentioned, the easiest would be if you are able to run a regression and find a seed that fails in your system. If not I have access to 20.09.009, I can try to find a failing seed with that simulator version.

eroom1966, 2022-02-21:
I just tried on 20.09 and the test you provided passed

eroom1966, 2022-02-21:
>  As I mentioned, the easiest would be if you are able to run a regression and find a seed that fails in your system

The only checks I generally run are individual tests or the ci_check prior to committing changes - I am happy to try your suggestion, but I would need detailed instructions in order to try this

silabs-mateilga, 2022-02-21:
@eroom1966 Using this tool, you should be able to generate a regression list, assuming you can run one of the supported options: https://github.com/openhwgroup/core-v-verif/tree/master/bin#cv_regress

Hopefully the documentation here is sufficient to get you going. What I've done in addition to this is to edit the /core-v-verif/cv32e40x/regress/cv32e40x_interrupt.yaml file so that it only runs the failing test, and up the number of runs to 100. That should give you at least one failing example.

If you're still not successful, I can try to find a instruction stream and seed combination that produces this fault in xcelium 20.09, but I think we have a higher chance of success if you are able to find a failing test with the regression-method.

eroom1966, 2022-02-21:
@silabs-mateilga 
I am unable to run this tool without some explicit guidance.

I do have cv_regress in my search path, so it is available, but I am completely unfamiliar with its usage.
In order that there is no possibility of misunderstanding, if you can provide the exact set of commands to run so that I can reproduce the issue I am happy to investigate.

I am afraid I do not really have the resource available to investigate the correct usage of tools I am not familiar with. I tried

> cv_regress -f /home/moore/git/bugs/core-v-verif/cv32e40x/regress/cv32e40x_interrupt.yaml
> cv_regress -f /home/moore/git/bugs/core-v-verif/cv32e40x/regress/cv32e40x_interrupt.yaml -p cv32e40x -s xrun

Unfortunately this does not seem to be the correct set of arguments

I really need a definitive set of instructions in order to reproduce your issue.

eroom1966, 2022-02-22:
I have no idea what vManager is, or where that comes from
I was able to run the command with the following output (hopefully correct)

> $ cv_regress --core=cv32e40x --file=cv32e40x_interrupt --vsif --lsf="-q himem" --sim=xrun --cov
> INFO:__main__:Reading regression: /home/moore/git/bugs/core-v-verif/cv32e40x/regress/cv32e40x_interrupt.yaml
> INFO:__main__:Rendering template: regress_vsif.j2 into: cv32e40x_interrupt.vsif
> INFO:__main__:Rendering template: regress_sve.j2 into: cv32e40x_interrupt.sve

I cannot find anything under core-v-verif called vManager
so it looks like this is a dead end

silabs-mateilga, 2022-02-22:
Try running the command with --sh in stead of --vsif. That should create a regression list that can be run in a shell.

eroom1966, 2022-02-22:
OK, that works - thx

eroom1966, 2022-02-22:
@silabs-mateilga 

can you please check the cv_regress command, I am seeing lots of runs but not all of them are the 
 TEST=corev_rand_interrupt_wfi_mem_stress
I am seeing many different tests executing

I thought the issue was a specific test ?

silabs-mateilga, 2022-02-22:
As I mentioned above, you need to edit the regression yaml file locally to run only the relevant test.

` What I've done in addition to this is to edit the /core-v-verif/cv32e40x/regress/cv32e40x_interrupt.yaml file so that it only runs the failing test, and up the number of runs to 100. That should give you at least one failing example.`

eroom1966, 2022-02-22:
thanks, now I have a list of 50 tests for 
TEST=corev_rand_interrupt_wfi_mem_stress

silabs-mateilga, 2022-02-22:
Hopefully you'll find one failure among them. I've seen on average 1/50 failing, so you might have to run it twice.

eroom1966, 2022-02-22:
I have managed to produce a reported failure, I am seeing if replaying it using the seed also fails

eroom1966, 2022-02-22:
Hmm, when I rerun with the seed applied manually - it does not fail, please check my logic

run number 3 fails
I grep for SEED in the logfile and I get 

> 3/xrun-corev_rand_interrupt_wfi_mem_stress.log:SVSEED set from command line: 155768169

I now run the following command line to try to reproduce 

> make test TEST=corev_rand_interrupt_wfi_mem_stress CFG=default SIMULATOR=xrun USE_ISS=YES COV=YES RUN_INDEX=3  GEN_START_INDEX=3  RNDSEED=155768169

but this does not fail, did I do something wrong ?

silabs-mateilga, 2022-02-23:
We've had some issues like this, but it's not really clear what does and does not cause the problem. What I'd suggest you try is the following:
Find the exact commands used in the regression generated script, both for generation (make comp_corev-dv..) and simulation (make test...)
run _make clean_all_ from cv32e40x/sim/uvmt
run the two commands from the regression, add "GUI=1 WAVES=1" to the second command

eroom1966, 2022-02-23:
@silabs-mateilga 
I had 1 fail in 100 tests, I have attached the log
based upon this log, can you tell me the exact command line I should execute to re-run so I can debug ?

I also include the script from cv_regress FYI

Thx
Lee

[xrun-corev_rand_interrupt_wfi_mem_stress.log](https://github.com/openhwgroup/core-v-verif/files/8122744/xrun-corev_rand_interrupt_wfi_mem_stress.log)
[cv32e40x_interrupt.txt](https://github.com/openhwgroup/core-v-verif/files/8122765/cv32e40x_interrupt.txt)

eroom1966, 2022-02-23:
@silabs-mateilga 

I don't understand, I looked at the part of the script which FAILED, this was iteration 18
I extracted the seed reported, and changed it from 'random', to the number used '926224011'
This is what I created below

pushd /home/moore/git/bugs/core-v-verif/cv32e40x/sim/uvmt > /dev/null
    make gen_corev-dv test TEST=corev_rand_interrupt_wfi_mem_stress CV_CORE=cv32e40x CFG=default COREV=1 SIMULATOR=xrun COMP=0 USE_ISS=YES COV=YES SEED=926224011 GEN_START_INDEX=18 RUN_INDEX=18 
popd > /dev/null

When I run this it does not exhibit the same reported mismatch, and runs to completion with a PASSED
any ideas what I could be doing wrong ?

Thx
Lee

silabs-mateilga, 2022-02-23:
So, from what I can tell from the log, your seed and index from your above does not match the log. Try the following:

`make clean_all`
`make comp_corev-dv gen_corev-dv TEST=corev_rand_interrupt_wfi_mem_stress CFG=default SIMULATOR=xrun USE_ISS=YES COV=YES RUN_INDEX=18 GEN_START_INDEX=18 RNDSEED=926224011`

`make test TEST=corev_rand_interrupt_wfi_mem_stress CFG=default SIMULATOR=xrun USE_ISS=YES COV=YES RUN_INDEX=18 GEN_START_INDEX=18 RNDSEED=926224011`

If that fails, rerun the last command with `GUI=1 WAVES=1`

eroom1966, 2022-02-23:
that did not fail, the fail was supposed to be at instruction 105901, running the commands above completed with a PASSED status
This is starting to sound like a race, or uninitialized variable.

I am going to try running the suite of random tests, with full instruction tracing turned on to see if I get a failure - I can then re-run and compare the full tracing

eroom1966, 2022-02-24:
@silabs-mateilga 
I captured the stdout from running a test, and it appears in the test 2 seeds are generated!
there is clearly a seed for program generation, and one for program execution

Is this our problem ?
I am applying the same seed for 2 different processes in standalone mode, but in the loop execution 2 different seeds are used for generation and execution

silabs-mateilga, 2022-02-24:
@eroom1966 Yes, this is likely the problem.
We had a discussion with @MikeOpenHWGroup yesterday, and he highlighted that the cv_regress diverges between the -sh and -vsif options in the use of seeds. The -sh version uses two seeds as you observed, while the -vsif version uses the same seed for both generation and simulation. He gave the impression this will change in the future, but maybe @MikeOpenHWGroup can comment further?

For you to recreate the issue I believe you will have to use the generation seed for the `make comp_corev-dv gen_corev-dv ` command, and the simulation seed for the `make test` command.

eroom1966, 2022-02-24:
Taking that approach I now have something I can now reproduce

eroom1966, 2022-02-24:
How do I increase the UVM_ERROR count - I want to proceed more than 5 errors

eroom1966, 2022-02-25:
@silabs-mateilga @MikeOpenHWGroup 
I have been investigating for the last few days, and I think I can see what is happening.
The problem we are looking at, is related to when a wfi instruction executes and the cpu goes into a halted state

I think we may have something missing in the detection of this state in the RVFI/RVFI and associated agents.
The specific scenario I see is when a wfi instruction is not immediately returned due to an active interrupt
This would also lead me to believe that there is potentially no directed test that addresses this scenario
I intend to put an assert for this specific condition, into the REF model, run the ci_tests and see if anything triggers the assert

Hopefully I can get some time with @MikeOpenHWGroup and work out what is happening here

Thx
Lee

MikeOpenHWGroup, 2022-02-27:
I took at look at this and can reproduce the error roughly one time in 30 sims on both the hash the issue was reported on (`fe9bdef`) and the current head of cv32e40x/dev (`f8316ae`).  The failing signature is the same summarized by Lee above.

> This would also lead me to believe that there is potentially no directed test that addresses this scenario

@eroom1966, that is almost certainly the case, and that would be consistent with a constrained-random verification strategy.  However, it might be worthwhile to review the functional coverage model to see if WFI is crossed with interrupts.

Question for @silabs-mateilga: where in the ENV hierarchy are those "RVVI" signals you posted (RVVI.insn, RVVI.pc, etc.)?  Admittedly, I am not a regular user of SimVision, but I could find everything else without difficulty.

eroom1966, 2022-02-28:
@MikeOpenHWGroup @silabs-mateilga 
The Imperas REF model in this repo/release was not exposing when it was in a halted state indication
I modified my local copy to indicate we are halted - this happens when we get a WFI and there are no pending interrupts, at this point the system locks-up, as the testbench does not assert any interrupts

I think we are missing a combination of indicating halted, and providing the interrupts when we are in a halted wfi state.

In the original issue, this would appear to be a synchronization issue when the halted state is not handled correctly

silabs-mateilga, 2022-02-28:
@MikeOpenHWGroup The rvvi signals can be found in the "iss_wrap" module; 
`uvmt_cv32e40x_tb.iss_wrap.cpu.state.insn`

eroom1966, 2022-02-28:
I think the problem we are seeing is related to 2 things.
1. I had missed the driving of the halt signal in the wrapper for the ISS/REF
2. the testbench is removing the deferint=0 value, which causes the wfi to lockup in a halted state

I probably need to have a shared desktop with you Mike, and whoever is now looking after the testbench

MikeOpenHWGroup, 2022-02-28:
I am available all morning today.  Suggest we bring in one or more of @silabs-mateilga, @silabs-hfegran or @silabs-robin as well.  I'll try to set something up on MM.

eroom1966, 2022-02-28:
@MikeOpenHWGroup - sent email, if you send a meeting invite I will get online straight away and show you what I know

eroom1966, 2022-02-28:
@MikeOpenHWGroup @silabs-oysteink @silabs-mateilga 
I have an interesting datapoint from our call earlier.
At the point where we enter the offending WFI instruction, I looked at the values in the MIP and MIE registers, and this is what I see

mie=0xA35B0800
mip=0x5CA40088

If I perform a bitwise AND of these 2 values the result is 0x00000000
this appears to be the reason why the REF model does not exit the HALTED state
The question is what does the RTL think these registers contain, because it does exit the HALTED state, indicating it thinks the result is not 0x00000000

Interestingly I see the MIP register transition
Info   mip 02000000 -> 5ca40088

IF the RTL bit in position 25 of the MIP is not cleared, then this would exit the HALTED state as the result would be 0x02000000

We should go ahead with the call tomorrow, and have a shared desktop again

silabs-mateilga, 2022-03-01:
@eroom1966 @MikeOpenHWGroup 
We have noticed a few interesting points on our sides as well, and will be attending the meeting.

eroom1966, 2022-03-01:
where is the meeting link ?

eroom1966, 2022-03-01:
Just checked - I have the email from Mike

silabs-mateilga, 2022-03-01:
@eroom1966 @MikeOpenHWGroup @Silabs-ArjanB 
I have created the task discussed in the meeting, links below:

[cv32e40x 466](https://github.com/openhwgroup/cv32e40x/issues/466)
[core-v-verif 1213(RVVI)](https://github.com/openhwgroup/core-v-verif/issues/1213)
[core-v-verif 1214(coverage)](https://github.com/openhwgroup/core-v-verif/issues/1214)

Please take a look and let me know if I've missed some important detail.

Silabs-ArjanB, 2022-03-01:
Hi @silabs-mateilga Can you also add a task to perform step-compare on the (to be added) rvfi_sleep signal?

[3 more replies]

closed 2023-01-30 by silabs-mateilga