openhwfoundation/core-v-verif #1195: CVXIF Agent Extended Architecture

silabs-robin, 2022-02-19:
I tried to work through the details of my previous diagrams, and either 1) I have over-engineered this, or 2) an XIF agent is really complex.
New detailed block diagrams are appended at the end of this issue.
@JeanRochCoulon , please let me know what you think of this proposed architecture of the CVXIF agent (and the proposed new per-isa-extension XIF reference model agent).
@MikeOpenHWGroup , you have more expertice on UVM than I, could you review these proposals?
@silabs-mateilga and @silabs-hfegran, any feedback from you too would be valuable.

The reference model agent is supposed to be unique per-isa-extension. So for instance, one agent for "P" extension via Spike, one agent for the current custom XIF instructions, etc (the details of said agent is a dont-care from this perspective).
If the rest of the diagrams are not self-explanatory, please just ask me questions; I have looked at it for so long now that I don't know what is obvious or not.
All the details might not be 100% correct, but I have run through all the major use cases and I think it is relatively complete, and we can discuss to figure out any weaknesses and so on.

One complicating factor is that, in the XIF, master/slave takes turn in who initiates (cause of mem interface).
Note, I have not decided whether the SB should check mem requests too (that can be decided later, unless someone foresees a significant infrastructure change?).
Note, I have not decided if the AgentXif or AgentRefmodel is responsible for holding inter-interface state relations (probably refmodel is best?), so please give suggestions.
Note, the driver wire is a bus (or a bus of buses), I just didn't want to repeat 9 wires as with the monitor wires.
Note, not shown in the diagrams, not all reqs/resps need to be fed to AgentRefmodel at all times (but it can) (e.g. slave mode doesn't need to see its own resps fed back to itself, but it needs to see mem resps).
Note, not all of these are equally valuable, so we should think about priority too.

Ps. Could we say "agent du refmodel" in pseudo-french? I like that "agent" comes first because of the way it groups different concepts in a sorted way.

Relates to https://github.com/openhwgroup/core-v-verif/issues/1150

Full agent:
[image]

Passive:
[image]

Active slave:
[image]

Active master:
[image]

MikeOpenHWGroup, 2022-02-20:
Also, @ZElkacimi should have a look at this.

MikeOpenHWGroup, 2022-02-20:
Thanks for continuing this discussion @silabs-robin.  I like where you are going with this as it follows "the UVM way" which will make it easier for individual projects to use the XIF Agent to suit their DUT's requirements.  A few points:
- Although "agent du refmodel" is an affront to both the English and French languages (and will get me in serious trouble with [l'office](https://www.oqlf.gouv.qc.ca/accueil.aspx)), I agree with the concept of keeping the reference model separate from the Agent.
- You do not explicitly state this, but your schematic hints that all stimulus should be provided by sequence_items, not generated by the Agent itself.  This is a very important consideration.
- To get started, let's keep the focus of this discussion on the `XIF sequence_item` and `XIF Agent`, not the Reference Model, Scoreboard and stimulus (sequences), as these are specific to each core.
- At some point, we may be able to define a "core agnostic" reference model and set of sequences, but if we focus on that too early we may lose focus on the short-term goal of verifying the X-interface implementation on the CVA6 and CV32E40X.

JeanRochCoulon, 2022-02-22:
Yes, I like this kind of discussion: defining the strategy before running !
I would be pleased to represent the agent as it is Today, may I ask you to share the source code of your "Full Agent" drawing ?

JeanRochCoulon, 2022-02-22:
Hello

From your architecture drawings, I have drawn the current XIF agent, as it is Today. Only the active slave is supported. Check is done by the CPU reference model. And only 3 interfaces are supported. Zineb is fine-tuning the sequences to be more generic, a PR will be submitted soon. To my mind, we should confirm the current agent implementation is compatible with the evolutions you/we are targetting.

[image]

For information, the review of your DVPlan comments are under review. We will come back to you soon.

Cheers

silabs-robin, 2022-02-25:
Thanks for your diagram of the current implementation.
It gave more insight into the situation.

I think we have slightly different intentions for the use of the agent (apart from just passive/active)?
You are checking two verification components against each other; the cpu-refmodel (written/modified by verification), and the extension-refmodel/sequences (also written by verification).
Comparing two verification components is maybe not the goal here(?), so the active slave is used just to exercise the XIF implementation?
One additional benefit of this active slave config, is that verification can start writing co-processor tests even _before_ an rtl coprocessor is ready (given that some refmodel is ready).
Am I correct in understanding that your intention for the xif agent is only to verify the xif implementation itself, and not the isa-functionality of a co-processor or anything else?

The active master configuration is what we envision for verifying the isa-functionality of a co-processor.
That is why a generic re-usable infrastructure for the agent could be beneficial.
(Also, the passive agent would come almost for free. While it is not the most useful, it can be used if a cpu-refmodel has not incorporated the isa-extensions yet or if one somehow have two different refmodels available and want extra checking.)

--------

Anyway... If we want to go for the agent architecture that I proposed before, here are proposal diagrams for an initial middle ground.

If this is the current setup:
[image]

Then maybe it would make sense to simply move the custom isa extension into an "extension-refmodel" agent?:
[image]

Then sometime later, the other xif interfaces can be added when desired:
[image]

(github doesn't easily allow me to share the visio source file)

silabs-robin, 2023-12-11:
Closed as stale. (Might resume if it becomes relevant again.)