openhwfoundation/core-v-verif #1225: make questa-test doesn't work

hasver, 2022-03-07:
I tried to run make questa-test as in the README file in the core directory 
The output is: 
make: *** No rule to make target `questa-test'.  Stop.

I explored the makefile and ran questa-custom instead of questa-test
The output is: 
/opt/corev-gcc/bin/../lib/gcc/riscv32-corev-elf/12.0.0/../../../../riscv32-corev-elf/bin/ld: cannot open output file /requested_csr_por/requested_csr_por.elf: No such file or directory
collect2: error: ld returned 1 exit status
make: *** [/requested_csr_por/requested_csr_por.elf] Error 1

MikeOpenHWGroup, 2022-03-07:
Hi @hasver, thanks for your issue.  I will need some additional information.  From your report my guess is that you are attempting to run simulations using the "core" testbench of the CV32E40P using Questa.  So your command line is probably something like this:
```
$ git clone https://github.com/openhwgroup/core-v-verif.git
$ cd core-v-verif/cv32e40p/sim/core
$ make questa-test TEST=hello-world
```
Please confirm that the above is correct.

Unfortunately, and to my surprise, there have not been any active members of the OpenHW Group using Questa for more than a year, so we have not been in a position to maintain the Questa targets in our Makefiles.  It would be a great contribution if you could help with that.  Like all open-source projects, the work only gets done if the community wants it to happen.   Also, be aware that the "core" testbench is not a production environment.  It is a demonstration platform only.

Looking at `core-v-verif/cv32e40p/sim/core/Makefile` I see there is no `questa-test` target.  So the README is out-of-date indeed.   By visual inspection, the following invocation _should_ work:
```
$ make questa-custom CUSTOM_PROG=hello-world
```

Let me know how that goes and we can work together to get Questa support updated.

leemango1998, 2022-04-12:
Hi @MikeOpenHWGroup, I found the reason why `make questa-custom` failed, that is because the `$(CUSTOM) ` is not defined in the `core-v-verif/cv32e40p/sim/core/Makefile` by default. So the riscv-gcc try to write the elf file into  `/hello_world/hellod_world.elf`, and the linker can't find it. So adding 
```
CUSTOM    = ../../tests/programs/custom
CUSTOM_PROG  ?= hello-world
```
into line[ 75](https://github.com/openhwgroup/core-v-verif/blob/1c8aa7c940229bf0528ae770bdc741a06645dcb1/cv32e40p/sim/core/Makefile#L75) will fix this issue.

And there is another issue in the makefile line [361](https://github.com/openhwgroup/core-v-verif/blob/1c8aa7c940229bf0528ae770bdc741a06645dcb1/cv32e40p/sim/core/Makefile#L361) . there should be a "space"  before the newline symbol `\` , otherwise the make fails.

MikeOpenHWGroup, 2022-04-12:
Thanks for this @leemango1998!   Would you be willing to generate a PR with the bug fix?  Our process for accepting PRs is documented in [CONTRIBUTING](https://github.com/openhwgroup/core-v-verif/blob/master/CONTRIBUTING.md).

leemango1998, 2022-04-13:
glad to contribute. But I am quite new to the PK flow. I will try it.

MikeOpenHWGroup, 2022-04-13:
Hi @hasver.  @leemango1998 has provided a fix (#1251) for this issue.  It is now merged onto the `cv32e40p/dev` branch.  Does that PR resolve this issue for you?