openhwfoundation/core-v-verif #1230: Verissimo Linter: UVM.2.1.4.2.8 - Create call arguments

MikeOpenHWGroup, 2022-03-20:
### Details
Verissimo is reporting several violations of `UVM.2.1.4.2.8 - Create call arguments`.  The defintion of this rule is:
```
Use 'this' as the second argument of a create call in order to provide a proper context for factory instance overrides.
Exception: Must use sequencer as the second argument when creating a non-virtual sequence in a test. The sequencer base class is uvm_sequencer.
```
Here is a typical example of the violating code: https://github.com/openhwgroup/core-v-verif/blob/cdf7ee911e6f2268cf98f2bc6e0f7e484a7f396d/cv32e40p/tests/uvmt/compliance-tests/uvmt_cv32e40p_firmware_test.sv#L136  The second argument to `create()` is a virtual sequence which is a descendent of `uvm_sequencer`.

### Steps to Reproduce
Check out the Verissimo SV/UVM Linter report of `cover-v-verif`, branch `cv32e40p/dev`, sha `cdf7ee9` [here](https://dvteclipse.com/core5verif-verissimo/1/reports/run/cdf7ee911e6f2268cf98f2bc6e0f7e484a7f396d/index.html)

### Additional context
AMIQ EDA uses core-v-verif as a publically viewable demo of their Verissimo SV/UVM linter tool.  Reports are available [here](https://dvteclipse.com/products/verissimo-linter?gclid=Cj0KCQiAjc2QBhDgARIsAMc3SqR0dbFLN6RGtBGYqibbyUNMYtiOg4ONOpUG23LWAq6i28EpVISWjJkaApB8EALw_wcB).

gabrielrad, 2022-03-21:
Hi Mike, thank you for the details, we're looking into this and we'll get back to you.

gabrielrad, 2022-05-03:
@MikeOpenHWGroup the reason why Verissimo reports a hit is because it considers the ***debug_vseq*** a virtual sequence and the exception is only applied for **_non-virtual_** sequences.

The rule definition for a virtual sequence is a class that extends from **_uvm_sequence_** and has **_uvm_sequence_item_** as type parameter.

Should we consider **_debug_vseq_** a non-virtual sequence? If not, what is the reasoning behind having a virtual sequencer as context?