openhwfoundation/core-v-verif #1233: core-v-verif fails to run on latest cva6 commit

zeeshandildar, 2022-03-21:
## Bug Title
Core-v-verif fails to run on latest cva6 commit https://github.com/openhwgroup/cva6/commit/8d893fb647d758bed5876307c64caa6adbc89f97 due to recent changes in cva6. 

### Type
* Functionally incorrect behavior

### Steps to Reproduce
1. clone the master branch using `git clone https://github.com/openhwgroup/core-v-verif` command
2. Set your RISCV_GCC path to the directory where your RISCV Tool-chain is installed e.g `export RISCV_GCC="/home/user3/tools/riscv/bin/riscv64-unknown-elf-gcc-10.2.0"` 
3. Set your RISCV_OBJCOPY path to the path where object copy is located e.g `export RISCV_OBJCOPY=/home/user3/tools/riscv/riscv64-unknown-elf/bin/objcopy` 
4. replace the `CVA6_HASH` in `{core-v-verif home dir}/cva6/regress/install-cva6.sh` with the latest commit hash `8d893fb647d758bed5876307c64caa6adbc89f97`
5. comment out `source ./cva6/regress/install-riscv-dv.sh` in `{core-v-verif home dir}/cva6/regress/dv-riscv-compliance.sh` as we don't need riscv-dv for this
7. Now run the compliance test from `core-v-verif` home directory by using command: `source cva6/regress/dv-riscv-compliance.sh`
8.  If then it gives the following error:
[image]
9. add `include "cf_math_pkg.sv" in `{core-v-verif home dir}/core-v-cores/cva6/common/submodules/common_cells/src/lzc.sv` before the module
10. Run again the compliance test as mentioned in step 7
Now you will see the following error:
[image]

### Additional context
Add any other context about the problem here.

MikeOpenHWGroup, 2022-03-21:
Hi @zeeshandildar, thanks for your issue.  We will look at this, but you should be aware that there is no requirement for the verification environment to run on the latest version of the RTL.  As you are probably aware, the CVA6 is in active development and it is natural for it to change in ways that affect the testbench.  This is why core-v-verif clones a specific version of the RTL - it gives the design team the freedom to make changes without breaking the testbench.  It is normal for us to be running regressions on a version of the RTL that is several commits behind the head.

Of course, as we get close to "verification complete", we need to pick a specific version, tag it and close coverage on that version of the RTL. We are many months away from that.

JeanRochCoulon, 2022-03-21:
Thanks for your feedback.

I add @ASintzoff in the loop

MikeOpenHWGroup, 2022-03-21:
Pull-request #1234 may (or may not) resolve this issue.

zeeshandildar, 2022-03-23:
Hi @MikeOpenHWGroup, thanks for your feedback. Actually we were working on the development of cva6 so, I thought it would be better to run tests on latest commit but it failed. However, I have created a pull request which solves this issue. Moreover, "AMO" tests are failing on both latest and old hash of cva6. I have also created [issue](https://github.com/openhwgroup/cva6/issues/840) for that too.

MikeOpenHWGroup, 2022-03-23:
> we were working on the development of cva6 so, I thought it would be better to run tests on latest commit

That is a reasonable, but invalid assumption.  It is normal for the RTL to be one (or more) steps ahead of the testbench, and it is important to give the RTL the freedom to make these changes without needing to worry about breaking the testbench.  This is why core-v-verif clones a specific hash of the RTL.

If you think that the testbench can or should be updated to use the latest RTL, please create a pull request to more the hash pointer in the Makefiles.

zeeshandildar, 2022-03-24:
As you have mentioned pull-request #1234 resolved this issue.

closed 2022-03-24 by zeeshandildar

zeeshandildar, 2022-03-25:
Closing this issue as it has been resolved by pull-request #1234