openhwfoundation/core-v-verif #1238: AMO tests are failing on cva6

zeeshandildar, 2022-03-29:
# Bug
Following riscv compliance tests are failing when run using core-v-verif. All of these tests are atomic operation tests:
1. rv32ua-amoxor_w
2. rv32ua-amoadd_w
3. rv32ua-amoor_w
4. rv32ua-amomin_w
5. rv32ua-amoand_w
6. rv32ua-amomaxu_w
7. rv32ua-amoswap_w
8. rv32ua-amominu_w
9. rv32ua-amomax_w

# Steps to Reproduce
1. clone the master branch using `git clone https://github.com/openhwgroup/core-v-verif` command
2. Set your RISCV_GCC path to the directory where your RISCV Tool-chain is installed e.g `export RISCV_GCC="/home/user3/tools/riscv/bin/riscv64-unknown-elf-gcc-10.2.0"` 
3. Set your RISCV_OBJCOPY path to the path where object copy is located e.g `export RISCV_OBJCOPY=/home/user3/tools/riscv/riscv64-unknown-elf/bin/objcopy` 
4. comment out `source ./cva6/regress/install-riscv-dv.sh` in `{core-v-verif home dir}/cva6/regress/dv-riscv-compliance.sh` as we don't need riscv-dv for this
5. Now run the compliance test from `core-v-verif` home directory by using command: `source cva6/regress/dv-riscv-compliance.sh`
6.  If then it gives the following error:
[image]
9. add `include "cf_math_pkg.sv" in `{core-v-verif home dir}/core-v-cores/cva6/common/submodules/common_cells/src/lzc.sv` before the module
10. Run again the compliance test as mentioned in step 5
11. It will run all the test listed in `{core-v-verif home direcotry}/cva6/tests/testlist_riscv-compliance-cv64a6_imafdc_sv39.yaml`.
Here is command line output after running riscv compliance tests
[image]

MikeOpenHWGroup, 2022-03-31:
Thanks for looking into this @ASintzoff.  However, the explanation, _depending on the [test bench] used, atomic handling is working well or not_, does not resolve the issue.

If there is something specific that @zeeshandildar needd to do (or not do) to resolve the error we should probably update a README and/or a launch script to avoid the issue.

zeeshandildar, 2022-03-31:
Thanks @ASintzoff for the update. As you said it depends upon the test bench whether atomic handling will work or not, so it is to mention that I have used `veri-core` test bench. So I will run it again using `veri-testharnes` as a test bench. Hopefully it will resolve this issue.

zeeshandildar, 2022-04-08:
After running using `veri-testharness` as a testbench all `AMO` tests got passed successfully.