openhwfoundation/core-v-verif #1244: Syntax Error

mattvlsi, 2022-04-02:
Reproduce
_$ cd $CORE_V_VERIF/bin
$ pip install -r requirements.txt_

**I'm getting some errors during installation requirements before starting to verification.
The error is:**
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-kqdo817d/constraint/setup.py'"'"'; __file__='"'"'/tmp/pip-install-kqdo817d/constraint/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-kqdo817d/constraint/pip-egg-info
         cwd: /tmp/pip-install-kqdo817d/constraint/
    Complete output (6 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-kqdo817d/constraint/setup.py", line 143
        except OSError, ex:
                      ^
    SyntaxError: invalid syntax
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

closed 2022-04-02 by mattvlsi

guojiazhuxi, 2022-11-23:
Excuse me, how is this solved?

MikeOpenHWGroup, 2022-11-23:
@ghost can you comment?

guojiazhuxi, 2022-11-23:
> @ghost can you comment?

The Constraint package in the Requirements.txt file is a Python2 package, not the Python3 package，Did you write wrong?

MikeOpenHWGroup, 2022-11-23:
@guojiazhuxi and @ghost can you have a look at pull-request #1505?  Thanks.