openhwfoundation/core-v-verif #1252: Update core-v-verif to support latest riscv-arch-tests infrastructure

M-Ijaz-10x, 2022-04-18:
Hi,

core-v-verif verification environment needs up-gradation in order to support the latest [riscv-arch-tests](https://github.com/riscv-non-isa/riscv-arch-test) infrastructure as core-v-verif is working at an older hash of riscv-arch-tests

**Why do we need to update**
The framework of the latest [riscv-arch-tests](https://github.com/riscv-non-isa/riscv-arch-test) is completely changed that is updated directory structure, updated macros, updated tests structure, etc. For testing any newly added extension in any of the core which uses core-v-verif as its verification environment, the latest compliance tests need to be run to test the functionality, (e.g. while adding support for bitmanip in the CVA6 core I needed bitmanip tests, which are present in updated riscv-arch-tests framework, but the core-v-verif CVA6 environment was not compatible with those tests).

**NOTE:**
As the current default branch [master] of [riscv-arch-tests](https://github.com/riscv-non-isa/riscv-arch-test) will be completely replaced with branch [riscof-dev](https://github.com/riscv-non-isa/riscv-arch-test/tree/riscof-dev) on **1st MAY 2022**. This transition essentially migrates the current framework to the RISCOF framework. This probably will break the core-v-verif environment and it will have to be updated. cc @MikeOpenHWGroup

M-Ijaz-10x, 2022-04-18:
I have made the following changes in order to use the latest riscv-arch-tests(master branch) in core-v-verif CVA6 environment

- Updated the hash of the riscv-arch-tests in the core-v-verif CVA6 environment
- Updated the testlist.yaml file with the current directory structure of the riscv-arch-tests
- While running tests on the core-v-verif environment the spike ISS simulation is malfunctioning, or more specifically the spike is using the pc for DATA_BEGIN section instead of CODE_BEGIN section, right now I am using a hack i.e --pc=’h8000_0000 but this needs to be fixed properly
- While logging the results of the spike and verilated model of CVA6, there is a mismatch of one instruction, i.e. the spike starts logging from pc value of 0x80000000 and the verilated model of CVA6 starts logging from the pc value of 0x80000004, hence the tests are failing in comparison of spike and model results, so either spike or verilator trace python script of CVA6 have to be updated - which needs to be fixed
- Updated the regex for ISA check in the cva6.py to identify bitmanip as a legal ISA extension
- Updated install-spike.sh script to use the latest version of the spike as the old version does not support bitmanip and many of the latest risc-v extensions. cc @MikeOpenHWGroup

MikeOpenHWGroup, 2022-04-18:
Hi @M-Ijaz-10x, thanks for creating this issue.

I am a little confused.  We met with @neelgala and @allenjbaum on March 1, 2022 to discuss using the updated RISCOF with core-v-verif.  We are aware that this will require an update on our end, but we were assured that using the compliance tests would not require significant changes to our verification environments.

Where are the changes you have made?  There are no pull-requests to this repository that I can see.

allenjbaum, 2022-04-18:
What I said was that *if* the open HW environment could run the existing framework, it should be able to run under the new one. This appears to be changes to make it work properly with the existing currently existing framework (as opposed to a much older one). 
Some of the changes above I can't explain, e.g. why spike is using DATA_BEGIN instead of CODE_BEGIN as a starting PC; that is particularly weird.

MikeOpenHWGroup, 2022-04-18:
Thanks @allenjbaum.  Where can I see these changes @M-Ijaz-10x?

M-Ijaz-10x, 2022-04-19:
> Thanks @allenjbaum. Where can I see these changes @M-Ijaz-10x?

I have made these changes locally, but I am updating the branch on the forked version of core-v-verif, I will share the link shortly

M-Ijaz-10x, 2022-04-19:
@MikeOpenHWGroup Please find all the above-mentioned changes [here](https://github.com/10x-Engineers/core-v-verif/tree/riscv-arch-tests-infrasturcture-upgradation), where I am running the latest riscv-arch-tests on CVA6.
My next step is to make it compatible with the [riscof-dev](https://github.com/riscv-non-isa/riscv-arch-test/tree/riscof-dev) infrastructure.

MikeOpenHWGroup, 2022-04-19:
Got it.  Just FYI, it might be better to start your work from the `cva6/dev` branch as this will be the most up-to-date.

@JeanRochCoulon, @ASintzoff do you have any comments?

allenjbaum, 2022-04-20:
Thanks for all this work. Please do complain if (realyl: When, I'm
guessing) you find something that is ambiguous, badly written, confusing,
or just broken.
We are trying to ensure this works as easily as possible for everyone.

ASintzoff, 2022-04-20:
> Got it. Just FYI, it might be better to start your work from the `cva6/dev` branch as this will be the most up-to-date.
> 
> @JeanRochCoulon, @ASintzoff do you have any comments?

Thanks @M-Ijaz-10x to work on this topic.

I would prefer you create a pull request to be able to provide detailed comments your commits.

Nevertheless, after a first reading, I have some general comments. Could you:
- create a new testlist yaml file and a new install script for riscv-arch-test instead of reusing the riscv-compliance ones
- avoid to use forked riscv-arch-test repository in install script. We prefer to use _official_ repositories.
- avoid to use git protocol in install script
- avoid to modify tool name environment variables (you can always overwrite them on your machine) or if needed, explain the underlying reasons in the commit message.

Silabs-ArjanB, 2022-04-20:
Hi @M-Ijaz-10x Without having looked at the actual changes you did, please also note that the core-v-verif parts for cv32e40x and cv32e40s were ported to the (then) current riscv-arch-test about 2 months ago (so that doesn't include RISCOF yet, but it should be close). It would be nice if the CVA6 and CVE4 verification parts don't diverge too much if not needed.

neelgala, 2022-04-20:
@M-Ijaz-10x how are you handling setting for compile-time macros like XLEN, TEST_CASE_1, etc ? These are macros that will need to be set based on the tests you run. If your testlist-format is like the one [here](https://github.com/openhwgroup/core-v-verif/blob/1ef5bda00d4b1271b74fc350aea5499fb90dfbae/cva6/tests/testlist_riscv-compliance-rv32ima.yaml) then you might want to handle setting those macros manually for each test. 

While for the current arch-tests this might work for now, since the macros are more-or-less fixed for each test. But future tests will include macros being set based on the yaml config of the DUT. So maybe including a node in the test-list yaml to capture the macros separately (like the way its done in the[ RISCOF Test-List yaml](https://riscof.readthedocs.io/en/stable/testlist.html) ) This will make the transition to RISCOF much easier, since then yuo will need to only convert the RISCOF test-list format to your test-list format and move completely to the framework of this repo there on.

MikeOpenHWGroup, 2022-04-20:
Hi @neelgala, the core-v-verif environments support signature dumping and this was used as part of our compliance signoff for v1.0.0 of CV32E40P.  While the signature writer is not otherwise used for the CV32E4 testbenches, it should still work.

Is there a document available detailing the operation of a RISCOF compliant signature writer and how a test-program running on the core is expected to use it?  The CV32E4 signature writers and related documentation is based on reverse engineering the signature writer in the original PULP-Platform RI5CY testbench.

pawks, 2022-04-20:
The TestFormatSpec in riscv-arch-test describes the signature file [here](https://github.com/riscv-non-isa/riscv-arch-test/blob/master/spec/TestFormatSpec.adoc#36-the-test-signature). RISCOF adheres to the same. The reference files checked in the riscv-arch-test repo can be used as an example([link](https://github.com/riscv-non-isa/riscv-arch-test/blob/master/riscv-test-suite/rv64i_m/I/references/add-01.reference_output)). Both SAIL and SPIKE(with `+signature-granularity=4`) implement signature dumping compliant to the same.

M-Ijaz-10x, 2022-04-25:
Thank you @MikeOpenHWGroup, @ASintzoff, and @neelgala for your suggestions. I think @fatimasaleem has covered all the aspects in her [PR#1254](https://github.com/openhwgroup/core-v-verif/pull/1254), with the changes made in the PR I am able to run all the latest riscv-arch-test on the current CVA6.

allenjbaum, 2022-04-26:
To be completely clear: is this with the existing framework,
or with the riscof-dev branch, which is what we're particularly interested
in

fatimasaleem, 2022-04-26:
> To be completely clear: is this with the existing framework, or with the riscof-dev branch, which is what we're particularly interested in

@allenjbaum It's with the riscof-dev branch.

allenjbaum, 2022-04-26:
Fantastic. We had no doubt you could do it; we were particularly interested
in any pain points you found so we can fix them

fatimasaleem, 2022-04-26:
Yes, for now, everything went smoothly. I will create an issue on `riscof-dev` if I found any pain points.

fatimasaleem, 2022-07-11:
[PR-1254](https://github.com/openhwgroup/core-v-verif/pull/1254) has been merged into `cva6/dev`, which addresses this issue and adds the support to run the latest riscv-arch-tests, now I think we can close this issue. Thanks.

allenjbaum, 2022-07-11:
Fantastic - Thanks much!