openhwfoundation/core-v-verif #1256: PMP vplan review, cv32e40s

silabs-robin, 2022-04-27:
### Task Outcome
The attached 40s vplan for PMP is reviewed, revised, accepted, and pushed to the repo.

### Location Information
The vplan is downloadable from here: [CV32E40S_UserMode.xlsx](https://github.com/openhwgroup/core-v-verif/files/8571732/CV32E40S_UserMode.xlsx).
A PR will later place the final file at `core-v-verif/cv32e40s/docs/VerifPlans/Simulation/privileged_spec`.

### Completion Criteria
The review procedure is completed (as per [here](https://github.com/openhwgroup/core-v-verif/blob/master/docs/VerifPlans/VerificationPlanning101.md#verification-plan-reviews)) and the finished file is in the repo.

silabs-robin, 2022-05-02:
_"Arrange for a review call for the verification plan"_
This can wait a bit, as we have more vplans coming up (PMP, CLIC, Zc), and it could be(?) more efficient to do them all at once or in close proximity.

Silabs-ArjanB, 2022-06-01:
[CV32E40S_UserMode-Arjan.xlsx](https://github.com/openhwgroup/core-v-verif/files/8812069/CV32E40S_UserMode-Arjan.xlsx)

MikeOpenHWGroup, 2022-06-09:
Is that #1287?

closed 2023-02-21 by silabs-robin