openhwfoundation/core-v-verif #1259: Core simulation failures using Questa with CV32E40P/E40X

leemango1998, 2022-05-04:
Hello, I noticed that the `Makefile` in `CV_CORE/sim/core` haven't been updated for three months. Actually the core simulation environments have broken for CV32E40P/E40X.` make questa-custom CUSTOM_PROG=hello-world` will not work now.
### Type
* Probably  Makefile bugs
### Steps to Reproduce
```
1. git clone https://github.com/openhwgroup/core-v-verif.git
2. cd core-v-verif
2. git checkout cv32e40x/dev
3. cd cv32e40x/sim/core
4. make questa-custom CUSTOM_PROG=hello-world
```

It shows `make: *** No rule to make target`. This makefile is originally a copy of `cv32e40p/sim/core/makefile`. Some modifies in `cv32e40p/sim/core/makefile ` is also needed by this makefile, like PR #1251. At this moment, I suddenly found that the PR #1251 was covered by the #1253  submitted later. The two PR's are in conflict actually. 
### Additional context

What I like to say is that the core simulation is as important as the UVM verification, especially for these who are not familiar with UVM, like me. And for cv32e40p, there is an example_tb that can simulate the core in [openhwgroup/cv32e40p](https://github.com/openhwgroup/cv32e40p). 

So in the following days, I will try to  firstly modify the `cv32e40x/sim/core/Makefile` to make it work with ''`make questa-custom CUSTOM_PROG=hello-world`'', then for cv32e40s. During this period,  I may ask for help here. I am sorry if I bothered you.

This work will be going in my fork[ cv32e40x/dev here](https://github.com/leemango1998/core-v-verif/tree/cv32e40x/dev).

leemango1998, 2022-05-04:
After some modifications, I can compile the core rtl soure codes and generate the hello-world.hex file. 

But when simulate the hex file, the questasim shows the uvm_fatal:  
`UVM_FATAL /home/mango/work/test/core-v-verif/cv32e40x/tb/core/mm_ram.sv(507) @ 1590ns: reporter [MM_RAM] out of bounds read from xxxxxxxx `. 
And here is the [simulate-log.txt](https://github.com/openhwgroup/core-v-verif/files/8618502/simulate-log.txt). 

At the simulation log, I noticed that there are some important warnings:
```
-- Optimizing module cv32e40x_tb_wrapper(fast)
** Warning: /home/mango/work/test/core-v-verif/cv32e40x/tb/core/cv32e40x_tb_wrapper.sv(82): (vopt-2685) [TFMPC] - Too few port connections for 'cv32e40x_core_i'.  Expected 57, found 55.
** Warning: /home/mango/work/test/core-v-verif/cv32e40x/tb/core/cv32e40x_tb_wrapper.sv(82): (vopt-2718) [TFMPC] - Missing connection for port 'data_rdata_i'.
** Warning: /home/mango/work/test/core-v-verif/cv32e40x/tb/core/cv32e40x_tb_wrapper.sv(82): (vopt-2718) [TFMPC] - Missing connection for port 'data_wdata_o'.
```
It looks like that the data ports of the core  are not connected to mem_ram! 

This is the data memory interface in `core-v-cores/cv32e40x/rtl/cv32e40x_core.sv`:

[image]

This is the instance of `cv32e40x_core` in `cv32e40x/tb/core/cv32e40x_tb_wrapper.sv`:

[image]

There are indeed missing `data_rdata `and `data_wdata` port connections!

There is another warning that I don't know what the impact is: 
`
** Warning: (vsim-3015) [PCDPC] - Port size (5) does not match connection size (12) for port 'clic_irq_id_i'. The port definition is at: /home/mango/work/test/core-v-verif/core-v-cores/cv32e40x/rtl/cv32e40x_core.sv(115).`

leemango1998, 2022-05-04:
Unfortunately, even after connecting the data bus with data memory(see my [fork](https://github.com/leemango1998/core-v-verif/commit/4674a3c9d78044f1c1e82919293d4760ef5406c6)), there are still write overflow issues:

`UVM_FATAL /home/mango/work/test/core-v-verif/cv32e40x/tb/core/mm_ram.sv(482) @ 19030ns: reporter [MM_RAM] out of bounds write to 00013c32 with 05b00000
`

Here is the [logfile.txt](https://github.com/openhwgroup/core-v-verif/files/8618845/logfile.txt). I am using pulp-gcc-ubuntu1804-20200913 with -march=rv32im and with the BSP files provided in cv32e40x/bsp.

There may be nothing I can do to solve the write overflow issue now. Your help will  be greatly appreciated.

MikeOpenHWGroup, 2022-05-04:
Hi @leemango1998, thanks for the bug report.  As we have discussed before, we have no current members of our projects using Questa on the CV32E40X or E40S cores, so the current status of their makefiles is unknown.  I am willing to work with you to get things working, but I do not have access to Questa.

Also, please keep in mind that the "core" testbenches are demonstration vehicles only and there is little interest within OpenHW to maintain the "core" testbench for the E40X and E40S. The "core" testbench has not been updated to support the more recent updates to the E40X, so there will be some work to do. If you are interested in the "core" testbench, I strongly recommend using the CV32E40P.

Lastly, I do not understand why you are seeing UVM messages from the core testbench.   By design, there should be no UVM code in the "core" testbench.

MikeOpenHWGroup, 2022-05-06:
Cool!  It would be a big help if you could create a pull-request to core-v-verif for this.

leemango1998, 2022-05-06:
I am not able to do so at present.  In [cv32e40x-sim](https://github.com/leemango1998/cv32e40x-sim), it's definitely not going to work with cv32e40x/bsp. So I used cv32e40p/bsp files. But in core-v-verif, even if I changed everything that may be wrong, like replacing ths BSP files into CV32E40P's,  the testbench will not work. There are still write overflow issues. I haven't figured out the real reason now.

closed 2023-04-13 by leemango1998