openhwfoundation/core-v-verif #1260: PMP vplan review, cv32e40s

silabs-robin, 2022-05-08:
### Task Outcome
The attached 40s vplan for PMP is reviewed, revised, accepted, and pushed to the repo.

### Location Information
The vplan is downloadable from here: [CV32E40S_PMP.xlsx](https://github.com/openhwgroup/core-v-verif/files/8647656/CV32E40S_PMP.xlsx)
A PR will later place the final file within `core-v-verif/cv32e40s/docs/VerifPlans/Simulation/`.

### Completion Criteria
The review procedure is completed (as per [here](https://github.com/openhwgroup/core-v-verif/blob/master/docs/VerifPlans/VerificationPlanning101.md#verification-plan-reviews)) and the finished file is in the repo.

Silabs-ArjanB, 2022-06-01:
[CV32E40S_PMP-Arjan.xlsx](https://github.com/openhwgroup/core-v-verif/files/8812053/CV32E40S_PMP-Arjan.xlsx)

silabs-robin, 2022-06-07:
We have had internal reviews and the vplan is being uploaded in a PR.
If no additional OpenHW-orchestrated review meeting is desired, then this issue can be closed.

MikeOpenHWGroup, 2022-06-09:
Is that #1288?

silabs-robin, 2022-06-13:
> Is that https://github.com/openhwgroup/core-v-verif/pull/1288?

Yes.

closed 2023-02-21 by silabs-robin