openhwfoundation/core-v-verif #1262: How to

athurwo, 2022-05-09:
# Template for CORE-V-VERIF Task Issues
Tasks are defined, assigned and tracked as GitHub Issues.  For obvious reasons,
the template for a task is very different from the template for a bug.

Please use this file as a template for creating Tasks to create or update any of
the documentation in this repository.
<br>**Hint**: click on the edit symbol (looks like a pencil) and then
edit the markdown source to create your issue.

## Task Title
A clear and concise description of the Task.  This can go in the titlebar.

### Task Outcome
Provide a brief description of what is expected.  For example, "Implement an
XYZ UVM Agent." If necessary point to a specification document.

### Background information
If necessary point to a specification in an outside document.

### Location Information
Specify the path, relative to the root of this GutHub project, of the code that
will be created and/or modified.

### Completion Criteria
Answer the question: how does the Assignee know they are done?

### Additional context
Add any other context about the problem here.

MikeOpenHWGroup, 2022-05-09:
Currently, the Imperas OVPsim is the only reference model supported by core-v-verif.  While we have investigated the use of Spike as a reference model using the [handcar](https://github.com/openhwgroup/core-v-docs/blob/master/verif/MeetingPresentations/CVA6/UsingHandcarCosimAPIs.pptx) API developed for Spike by the FORCE-RISCV project, for a variety of reasons, these efforts have not generated sufficient interest within the community to complete the integration of Spike.

There is a prototype wrapper for handcar for the [cv32e40x](https://github.com/openhwgroup/core-v-verif/blob/master/cv32e40x/tb/uvmt/uvmt_cv32e40x_rvvi_handcar.sv).  If you would like to complete this, that would be a terrific contribution.

alpkoyun, 2023-01-03:
@MikeOpenHWGroup Hello, as I understand the rest of the system is independent of the ISS as long as the RVVI interface provides the necessary data and the only part that needs to be set up is the RVVI right? I would like to try to implement the Spike, I feel like I understand the verification methodology but I have no idea about which file is called where and the connections between them . Do you have a diagram or any kind of document about the system hierarchy that you can provide? Thank you.

MikeOpenHWGroup, 2023-01-03:
> as I understand the rest of the system is independent of the ISS as long as the RVVI interface provides the necessary data and the only part that needs to be set up is the RVVI right?

That is (almost) completely accurate as of today.  As you may know, we have switched all of our verification environments for the CV32E4 cores to use the new ImperasDV Reference Model and the interface to that RM is RVVI. You can find the specification for RVVI [here](https://github.com/riscv-verification/RVVI).  The top-level README provides some useful documentation for connecting an RTL model of a core to an RM using RVVI.  If you like, you can use that as a starting point.

I do not believe this strategy will work with Spike because Spike is instruction-cycle-accurate and the RTL is clock-cycle-accurate.  The implication here is that asynchronous events such as interrupts, debug-requests and memory bus errors will be handled differently by the RTL model and the RM.  The ImperasDV RM is specifically implemented to resolve this problem, but Spike is not.

If you are going to use Spike and want to handle asynchronous conditions, I would suggest the [Step-and-Compare-2.0](https://docs.openhwgroup.org/projects/core-v-verif/en/latest/cv32_env.html#step-and-compare-2-0).  We used this successfully for the CV32E40P, E40X and E40S, and although all have switched to RVVI+ImperasDV it is still a very solid methodology and I believe that you could be successful using Spike as a reference model with Step-and-Compare-2.0.

This is a BIG topic - too big for a discussion in a GitHub Issue.  My email is on my GitHub profile - send me a note and we can arrange a discussion.

alpkoyun, 2023-01-05:
I have looked at the documents and the handcar. As you said the topic is too big to discuss on a github issue but I need some quick clarification to see the expected size of the issue, I would be grateful if you clarify them. The original step and compare did not include RVVI, RFVI etc and the step and compare 2.0 introduced these interfaces with some modifications on the stepping and comparing algorithm I assume.
> We used this successfully for the CV32E40P, E40X and E40S, and although all have switched to RVVI+ImperasDV 

The current master branch of the environment uses the RVVI interface which, as I understand was introduced with step and compare 2.0. . Since you did not classify this as step and compare 2.0, does this mean the current stepping and comparing algorithm similar to the original one rather than the step and compare 2.0? 

The original step and compare assumes that model is cycle accurate but the  step and compare 2.0 assumes the model is instruction accurate.  Would this mean step and compare 2.0 algorithm does not require any change or only requires a minuscule amount of change to introduce a (handcar or spike model)?

MikeOpenHWGroup, 2023-01-05:
Ah, I see I was not clear.   Step-and-Compare does not used RVVI and ImperasDV.  Step-and-Compare (both 1.0 and 2.0) used bespoke Tracers from the RTL (each one was unique to it's specific core) and OVPsim as the reference model.

None of our environments assume that the reference model is cycle accurate.  It was the responsibility of the testbench to handle the differences between the cycle accurate RTL and instruction cycle accurate reference model.  So there were two significant problems:
1. The Tracer interface was unique for each core.
2. The cycle-to-instruction cycle deltas were different for each core.

Moving to RVVI solves the first problem.  If every core supports RVVI as a Tracer interface then connecting the Tracer to the verification environment is greatly simplified.  Step-and-Compare 2.0 was the first step in that transition.

Adopting ImperasDV solves the second problem.  Unlike OVPsim, which is an Instruction Set Simulator, ImperasDV is a true Reference Model.  ImperasDV is able to handle the cycle-to-instruction cycle deltas which means that the verification environment does not need to.

Because Spike is an ISS, it does not handle the cycle-to-instruction cycle deltas.  So I recommend using the S&C2.0 strategy.  This will mean the deltas will need to be handled in the verification environment and will be unique for each core.

maolei-l, 2023-05-15:
> Because Spike is an ISS, it does not handle the cycle-to-instruction cycle deltas. So I recommend using the S&C2.0 strategy. This will mean the deltas will need to be handled in the verification environment and will be unique for each core
@MikeOpenHWGroup 
well, I understand what you're saying about the difference between ImperasDv and spike.
So I decided to go with s&c2.0 strategy.
However, cove-v is an ongoing project, which is not tagged with the corresponding version number.
Can you provide a better version for spike?

It will be clearer if the development history of ocre-v is added in the readme (with the corresponding git version number). Perhaps it can be divided by the development of s&c:
1. s&c1.0
2. s&c2.0
3. rvvi+imperasDv

MikeOpenHWGroup, 2023-05-15:
Hi @maolei-l, we do not track specific revisions of CORE-V-VERIF, except as it relates to a specific "core-under-verification".  As each core achieves its verification goals we tag the specific version of CORE-V-VERIF that was used to verify a specific core.

We have no plans to implement specific releases of CORE-V-VERIF based on other criteria.

maolei-l, 2023-05-24:
Hi @MikeOpenHWGroup 
Ok, I'll analyze the history of repo updates to find the most suitable code.
Does Step-and-Compare 2.0 include rvfi interface?

MikeOpenHWGroup, 2023-05-24:
Yes, all CORE-V cores developed by the OpenHW Group use some version of RVFI.  The best source of documentation for that is in the CV32E40S User Manual ([link](https://docs.openhwgroup.org/projects/cv32e40s-user-manual/en/latest/rvfi.html)).

S&C 2.0 uses RVFI, but it could be made to work with almost any tracer you have.  The most up to date documentation for the implementation of S&C 2.0 is [here](https://docs.openhwgroup.org/projects/core-v-verif/en/latest/cv32_env.html#step-and-compare-2-0).

maolei-l, 2023-05-24:
Then can I classify it in such a simple way?  
1. s&c1.0: tracer
2. s&c2.0: rvfi (rvfi is similar to tracer, but made a common interface)
3. Current: rvvi+Imperas Dv.

So, if I want to refer to s&c2.0, I just need to find the repo before updating imperas dv?

MikeOpenHWGroup, 2023-05-24:
Yes, that is reasonably accurate.  The tracer used for S&C1.0 was "ad-hoc" and not well documented, so your best bet is to look at RVFI.  The differences between RVFI and RVVI are not significant for single-hart, in-order cores.

maolei-l, 2023-05-25:
OK, thanks.
If I simply use iss instead of imperas dv. Can I use rvvi?
Because in the paper, s&c 2.0 only uses rvfi.
[image]

MikeOpenHWGroup, 2023-05-25:
Whether you use RVFI or RVVI is or any other interface/API to connect to the Spike ISS is your decision.  For the CV32E4* class cores the distinction between RVFI and RVVI is not significant. If you have a more complex core that supports OoO execution or can retire multiple instructions in a single cycle then you'll want RVVI.

djvra, 2025-08-14:
Is the RTL for CV32E40P up to date? When I look at this https://github.com/openhwgroup/cv32e40p/tree/master/rtl I see some file changes.

MikeOpenHWGroup, 2025-08-14:
> @MikeOpenHWGroup Is the RTL for CV32E40P up to date? When I look at this https://github.com/openhwgroup/cv32e40p/tree/master/rtl I see some differences.

Hi @djvra, this is a new question with a different subject and should be asked in a separate Issue.

Also, if I have answered your original question (How to run ISS with Spike), please close this Issue.

MikeOpenHWGroup, 2025-08-14:
Sorry, @djvra, you are right, this Issue was opened by @athurwo.  Hi @athurwo , we have not heard from you for > 2 yrs, so I am closing this Issue.