openhwfoundation/core-v-verif #1268: uvma_rfvi: non compliant LRM SystemVerilog code

ZElkacimi, 2022-05-13:
With VCS (which is picky regarding LRM SystemVerilog), I get the following compilation error
 “ Following verilog source has syntax error :
Select on function call uvma_rvfi_instr_mon.sv
 token is ']' 
mon_trn.csrs["dcsr"].get_csr_retirement_data()[3]);

file with the syntaxe error is at: https://github.com/openhwgroup/core-v-verif/blob/master/lib/uvm_agents/uvma_rvfi/uvma_rvfi_instr_mon.sv.

To reproduce the error, compile the following example:

**class base_seq;**
    
 **function bit [5:0] testsyntx();
           return (5'b00100);
 endfunction**

 **endclass**

**class seq_c extends base_seq;
   bit zi;
function new(string name="seq_c");
      super.new(name);
   endfunction : new**
   
**task test_sytax();
      zi = testsyntx()[2];
   endtask**

**endclass**

use the following command: vcs  -sverilog file.sv

Error message: "Error-[SE] Syntax error
  Following verilog source has syntax error :
  	Select on function call
  "not_lrm_compliant.sv", 28: token is ';'
        zi = testsyntx()[2];"

Same error with the option :  -xlrm

**_Possible solution:_**
Use an intermediate variable:
 bit [5:0] inter_var = testsyntx();
 zi = inter_var[2];

MikeOpenHWGroup, 2022-05-13:
Hi @ZElkacimi.  VCS is doing the right thing - this code construct is not compliant with the LRM.  Did it compile/run with another simulator?  In any case, it must be fixed.  Please create a pull request.

ZElkacimi, 2022-05-17:
Hello @MikeOpenHWGroup

I don't have the opportunity to  compile/run with another simulator, but since it's in lib/ directory then I suppose that it works with another simulator. I have created a PR for a simple fix, please check it to see if it's the best one to do.

Thanks

MikeOpenHWGroup, 2024-04-05:
Hi @ZElkacimi.  As this Issue has not received any comments in almost 2 years and we do not believe there is an issue with the uvma_rvfi, I am closing this issue.  Please feel free to reopen if you have further questions.

closed 2024-04-05 by MikeOpenHWGroup