openhwfoundation/core-v-verif #1294: [TASK] Add riscv-arch-test suite to existing CVA6 core-v-verif CI (32-bit and 64-bit)

spidugu444, 2022-06-10:
### Is there an existing core-v-verif task for this?

- [X] I have searched the existing task issues

### Task Description

Adding riscv-arch-test suite for riscv cva6 environment.

Steps:
1.Please find the [setup](https://github.com/openhwgroup/core-v-verif/pull/1254) used to bring up cva6   riscv-arch-tests with the riscof-dev infrastructure
2.Need to update [spike](https://github.com/riscv-software-src/riscv-isa-sim/releases/tag/v1.1.0) version 1.1.0.

### Description of Done

perform CI run

jquevremont, 2022-09-16:
PR submitted in Thales internal CI.

jquevremont, 2022-10-28:
@ASintzoff provided feedback. Waiting for the fix.

jquevremont, 2023-01-20:
Investigating too long simulation time.
@ASintzoff: need to check the trigger of the end of the simulation (instead of waiting for the timeout).

jquevremont, 2023-03-14:
Now solved.

closed 2023-03-14 by jquevremont