openhwfoundation/core-v-verif #1298: [TASK] Provide VPTOOL to write Verification plan

JeanRochCoulon, 2022-06-10:
### Is there an existing core-v-verif task for this?

- [X] I have searched the existing task issues

### Task Description

Provide VPTOOL to edit the verification plan
VPTOOL windows organization is aligned on OpenHW Group verification framework.
Restore/save database is in Yaml.

### Description of Done

Edit CV-X-IF verification plan with VPTOOL
Commit the yaml generated CV-X-IF database in github

zchamski, 2022-07-12:
PRs to be linked to this task:

* #1331 (VPTOOL initial release)
* #1332 (VPTOOL documentation fixes)

Yet to come:

* Persistent storage in Yaml format
* CV-X-IF verification plan in Yaml

jquevremont, 2022-09-16:
2022-09-16: work in progress. 2 limitations detected.

JeanRochCoulon, 2022-09-22:
Following the elaboration of the first dvplan (FRONTEND) with VPTOOL, VPTOOL need this improvements:
- Fill-up the DVPLAN intro chapter named "getting start verification", could be a copy from README
- In the md file, PASS/FAIL Criteria is a number, to be replaced by the corresponding ASCII field.
- Add the design document version, the DVPLAN author and the VPTOOL version. To be exported in md file.
- Keep the database legacy when updating VPTOOL
- Add SPDX in md file
- vp_config.py is currently located in FRONTEND directory, VPTOOL must use a common vp_config located in tools/vptool/vptool directory
- Label Name must be F000SF001I002 like Feature000Subfeature001Item002
- (more long term action) be able to save/restore in YAML format

jquevremont, 2022-10-28:
André: The task has progressed since last month with users' feedback. First usable version targetted e/o 2022.

zchamski, 2022-11-04:
* [DONE] In the md file, PASS/FAIL Criteria is a number, to be replaced by the corresponding ASCII field.
  * Fixed for all fields with enumerated or bitmap-based values: `Pass/Fail Criteria`, `Test Type`, `Coverage Method`, and `Applicable Cores`.
  * see PR core-v-verif#1445.

zchamski, 2022-11-17:
Status update after PR https://github.com/openhwgroup/core-v-verif/pull/1493 was merged:
>   * *vp_config.py is currently located in FRONTEND directory, VPTOOL must use a common vp_config located in tools/vptool/vptool directory*

**DONE**
>   * *Label Name must be F000SF001I002 like Feature000Subfeature001Item002*

**DONE**
>   * *Remove dead variable/code*

***"Mostly done"*** (still to do: detailed cleanup, `pylint` metrics improvements)

Remaining short-term TODOs:
  * *Fill-up the DVPLAN intro chapter named "getting start verification", could be a copy from README*
    * affects static content in `cva6/docs/VerifPlans/source`, does not affect `VPTOOL` code per se
    * does not affect vplan database content 
  * *Add the design document version, the DVPLAN author and the VPTOOL version. To be exported in md file.*
    * new features to be added to `VPTOOL` code
    * currently under study
    * does not affect vplan database content
  * *Add SPDX in md file*
    * new feature to be added to `VPTOOL` code
    * does not affect vplan database content

Longer term TODO:
* *(more long term action) be able to save/restore in YAML format*

JeanRochCoulon, 2022-11-17:
From this morning meeting, these topics should be addressed:
- remove the comments and link to coverage information from markdown when empty
- Add the reference specification with url and version/hash
- Find a way to link to the right chapter of the reference spec
- Push PR into vptool/dev

MikeOpenHWGroup, 2022-11-17:
> Push PR into vptool/dev

I _like_ this idea!   Recommend that we first merge `cva6/dev` to master and then create a new branch called `vptool/dev` from master.

zchamski, 2023-01-20:
DONE:

* Add the reference specification with url and version/hash
* Find a way to link to the right chapter of the reference spec

Still TODO:
* remove the comments and link to coverage information from markdown when empty
* Push PR into vptool/dev

zchamski, 2023-02-24:
DONE:

* Yaml storage support: added in PR https://github.com/openhwgroup/core-v-verif/pull/1646.

zchamski, 2023-03-14:
PR #1646 was merged 24 Feb 2023.