openhwfoundation/core-v-verif #1305: Reference Model: Default mtvec_mask=0xffff_ff01 incorrect (should be 0xffff_ff81)

eroom1966, 2022-06-16:
@silabs-hfegran
Reference
https://github.com/openhwgroup/core-v-verif/issues/1285

Could you please provide instructions on how to reproduce this difference between the RTL and RM

Silabs-ArjanB, 2022-06-20:
@eroom1966,

>>If I change the default for the mtvec_mask from 0xffff_ff01 to 0xffff_ff81, 

We need that change anyway, even if a test starts failing.

@silabs-hfegran  The related code looks okay to me in the RTL and in RVFI. Is there some glue in core-v-verif perhaps interfering with this?

Silabs-ArjanB, 2022-06-20:
Hi @eroom1966 Is it possible to have a screen sharing session where you show this issue today?

eroom1966, 2022-06-20:
@Silabs-ArjanB today is difficult - we could schedule for early tomorrow, or I could provide exact details how to reproduce ?

Silabs-ArjanB, 2022-06-20:
Hi @eroom1966 Let's try early tomorrow then. I will send a meeting invite; just propose a different time if it is too early.

silabs-mateilga, 2022-06-20:
@eroom1966 We don't have problems writing this csr on our simulations. If you can provide details on how to reproduce, maybe I can have a look and figure out the difference before the meeting.

eroom1966, 2022-06-20:
aaah, apologies - attempting to re-create all my steps now. ....

Silabs-ArjanB, 2022-06-21:
@eroom1966 As far as we can see the issue you describe above does not exist. The PR with ISS and RVVI fixes was merged yesterday (see https://github.com/openhwgroup/core-v-verif/pull/1307). Could you please check if the issue you mention here still exists on (out of the box) core-v-verif after ÅR 1307?

closed 2022-06-21 by eroom1966