openhwfoundation/core-v-verif #1309: Cover cv32e40x issue #325 in a regression test

silabs-mateilga, 2022-06-20:
https://github.com/openhwgroup/cv32e40x/issues/325

This bug does not pop up in regression, this needs to be covered.

eroom1966, 2022-08-05:
I can reproduce this issue running cv_regress with the ImperasDV flow, I am not sure why your (current) testbench does not flag this as an error

silabs-hfegran, 2022-11-18:
@silabs-mateilga : What is the status on this? can this be closed?

silabs-mateilga, 2022-11-18:
@silabs-hfegran I am not aware that any effort has been made to create an assertion and/or directed test to cover this.

silabs-robin, 2023-12-11:
> Issue has been fixed for X_EXT = 0. The issue will remain open to look at it again once we start fixing the X interface (X_EXT = 1) (and therefore its title has been prefixed with [XIF]).