openhwfoundation/core-v-verif #1325: Test: working with issues

MikeOpenHWGroup, 2022-06-30:
Hi Joe  and Maarten, this issue is a test of the following:

1. A check to see if I can assign an issue to you (I can).
2. A check to see if you receive an email from GitHub now that you've been assigned an issue.

Please reply to this issue if you do receive the email.

joecircello, 2022-06-30:
Mike,

Yes, I got a notification email indicating issue #1325<https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fopenhwgroup%2Fcore-v-verif%2Fissues%2F1325&data=05%7C01%7Cjoe.circello%40nxp.com%7C6f8d727f4fc646987f4f08da5a2d6fde%7C686ea1d3bc2b4c6fa92cd99c5c301635%7C0%7C1%7C637921448447688029%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=sGa55iKrcFZjCPRFoj%2BgtomCJEfCJIf81mf7uV3CzHY%3D&reserved=0> had been assigned to me.

  *   Joe

Joe Circello
Technical Fellow, Chief Engineer
NXP Semiconductors
1300 North Alma School Road, MD AZ50-CH230
Chandler, AZ  85224
United States of America
Email: ***@***.***

***@***.***

Caution: EXT Email

MaartenArts, 2022-07-01:
Hi Mike

Github issue assignment email received in good order, thx

Best regards
Maarten