openhwfoundation/core-v-verif #1326: Imperas ISS: Setting cycle_undefined=T removes mcycle csr

silabs-mateilga, 2022-06-30:
When setting the ISS override cycle_undefined=T, the mcycle csr is removed. This is not correct behaviour, as it is only the cycle csr(user mode mcycle alias) that is removed. Also, the default setting for the parameter should be "cycle_undefined=T" for the cv32e40s.

### Steps to Reproduce
1. [URL to branch that exhibits the issue.](https://github.com/openhwgroup/core-v-verif/tree/cv32e40s/dev)
2. Command line: make test TEST=cv32e40s_csr_access_test

### Additional context
Note that the test fails either way, but with the ISS parameter set the test fails at order 4, e.g. almost immediately, with a mismatch where the ISS decodes a csrrw of mstatus as illegal.

eroom1966, 2022-07-08:
@silabs-mateilga 
Back from vacation and in the process of catching up.

my understanding is that the CV32E40S supports user mode
are you saying there are no user mode accessible CSR registers ?

eroom1966, 2022-07-08:
Ignore previous comment
I now understand what you are saying - access to the cycle register in USER mode, is dependant upon the setting of mcounteren[CY]
if mcounteren[CY] is 0, then this should trap, going to take a look at your testcase in more detail

silabs-mateilga, 2022-07-08:
Was just about to reply, but it seems you caught up before I finished.. FYI I will be on vacation the first two days of next week.

eroom1966, 2022-07-08:
@silabs-mateilga 
in your infrastructure, which file do you use to set/clear the override ?
cycle_undefined=T

silabs-mateilga, 2022-07-08:
In /lib/uvm_agents/uvma_rvvi_ovpsim/uvma_rvvi_ovpsim_agent.sv, approx line 137.

` if (cfg.core_cfg.core_name == "CV32E40S") begin
     $fwrite(fh, $sformatf("--override root/cpu/Smstateen=T\n"));
     $fwrite(fh, $sformatf("--override root/cpu/cycle_undefined=T\n"));
   end`
   
You should be aware that there may or may not be an issue with the way this system generates these, there was, and I'm not sure if it's fixed. The safest way to run this is to do a make clean_all for every run.

eroom1966, 2022-07-08:
If you are setting cycle_undefined=T, I think this is a mistake, because mcycle is not undefined, and cycle (in user mode) should only generate an exception when mcounteren[CY]=0

I still think there is something missing in your instructions as I cannot reproduce a register miscompare, unless what you are saying is that when setting cycle_undefined=T causes the miscompare, that makes total sense as there would be a miscompare of mcycle in machine mode

mcounteren powers up as 0, so a read from cycle in user mode, should cause an exception I believe

we can share a desktop to take a look if you wish ?

silabs-mateilga, 2022-07-08:
I made the mistake of not noting the hash where I found the issue, and things have changed. I'm trying to get a clear reproduction again now. I'll get back to you when I've sorted this out.

silabs-mateilga, 2022-07-08:
@eroom1966  to make it clear, the issue is that when trying to access mcycle in machine mode, the ISS decodes the instruction as illegal, when the setting is set. In other words, legal accesses to "mcycle" are deemed illegal by the ISS when the alias "cycle" is set as undefined.

Reproduce, attempt two:
Still, go to the latest hash of cv32e40s/dev in core-v-verif.
In /lib/uvm_agents/uvma_rvvi_ovpsim/uvma_rvvi_ovpsim_agent.sv, starting line 135, there is an if case for 40s ovpsim overrides. Modify it by adding the cycle_undefined, as shown below.

    if (cfg.core_cfg.core_name == "CV32E40S") begin
    $fwrite(fh, $sformatf("--override root/cpu/Smstateen=T\n"));
    $fwrite(fh, $sformatf("--override root/cpu/cycle_undefined=T\n"));
    end

Now, the _make test TEST=cv32e40s_csr_access_test_ should stop on order 4, with an mcause mismatch. Please let me know if this doesn't work.

eroom1966, 2022-07-08:
@silabs-mateilga 
the reason you are getting an illegal instruction is because you have used the setting 
cycle_undefined=T

Overriding the default value of False seems wrong to me.
This is how I interpret your design, please correct me

Firstly, there is only **one** register which holds the cycle count, it is accessed as follows in the different modes.

mcycle register at address 0xB00, only accessible in MACHINE mode

cycle register at address 0xC00, which can be read in USER mode, without generating an illegal instruction **if** mcounteren[CY]=1
**if** mcounteren[CY]=0, a USER mode access of cycle should generate an illegal instruction

is my interpretation incorrect ?

eroom1966, 2022-07-08:
If you do not override the cycle_undefined from the default setting of False, can you tell me where the difference occurs between the RTL and the ISS - because I see no difference when I execute

silabs-mateilga, 2022-07-08:
Resolved as documentation issues.

Silabs-ArjanB, 2022-07-14:
@silabs-mateilga Leaving this open until it has been shown that in machine mode mcycle can be accessed and cycle cannot be accessed.

eroom1966, 2022-07-14:
Hi @Silabs-ArjanB 
I think this is effectively resolved, but wanted to clarify when you said 'not a documentation issue'

I was specifically referring to version 0.2.0 here
https://docs.openhwgroup.org/projects/cv32e40s-user-manual/en/0.2.0/control_status_registers.html?highlight=mcounteren#machine-counter-enable-mcounteren

bit 0 of this register 
RW | CY: **cycle enable for user mode**.

This was the documentation error I was referring to, there is a RW bit to enable the non-existant cycle register
In fact I presume this is also the case for instret

Silabs-ArjanB, 2022-07-14:
Hi @eroom1966 , understood now. The mcounteren documentation issue has been solved in a later (0.4.0) release indeed. cycle and instret are treated in similar manners indeed. @silabs-mateilga just confirmed that mcycle/cycle behave as documented in both ISS and RTL on the 40S.

closed 2022-07-14 by Silabs-ArjanB