openhwfoundation/core-v-verif #1327: Imperas ISS: PUSH and POP type instructions marked as illegal instructions

silabs-oysteink, 2022-07-01:
## Bug Title
PUSH and POP type instructions marked as illegal instructions

### Type
* Functionally incorrect behavior

### Steps to Reproduce
1. Use my fork of core-v-verif: git@github.com:silabs-oysteink/core-v-verif.git
2. Checkout my branch "silabs-oysteink_zc-decode"
3. Check out CV32E40X hash 47d4a57b9e27363ff152c364324bdbbf91a06367
4. Modify cv32e40x/rtl/cv32e40x_core.sv to set localparam ZC_EXT= 1
5. make clean test TEST=test_zc CV_CORE_PATH=<path_for_cv32e40x_from_above> USE_ISS=1 CV_CORE=cv32e40x

My branch of core-v-verif is modified to enable ZC_EXT for the ISS, and a directed test "zc_test" is added with manually encoded Zc instructions.

The image below shows the test output. After running through some initial Zc instructions, there is a mismatch on the first PUSH instruction. If one comments out the first failing instruction, the next push/pop type instruction will also mismatch on mcause. 

[image]

MikeOpenHWGroup, 2022-07-22:
@silabs-hfegran, this issue looks like it has gone stale.  What is its status?

silabs-hfegran, 2022-07-25:
> @silabs-hfegran, this issue looks like it has gone stale. What is its status?

@MikeOpenHWGroup,
I am uncertain about the current status of this issue - perhaps @silabs-mateilga or @silabs-oysteink can give an update here next week?

eroom1966, 2022-11-04:
I am still assigned on this issue - is there something for me to look at ?
I think this is probably pending fixes to the RVFI interface to drive multiple destination registers in a single retirement ?

silabs-mateilga, 2022-11-04:
@eroom1966 I think this issue is deprecated now, I'll get Øystein to close it, and we'll open new issues if there is still something here. That said, the RVFI interface is expanded to drive multiple registers, and has been for weeks now. The signals are the set of rvfi_gpr_* found [here](https://github.com/openhwgroup/cv32e40s/blob/master/bhv/cv32e40s_rvfi.sv#L315).

silabs-oysteink, 2022-11-04:
This issue is likely out-of-date. I'll close it, and if the issue comes back we can make a new or reopen this one.

closed 2022-11-04 by silabs-oysteink