openhwfoundation/core-v-verif #1330: Update CSR-access test

silabs-hfegran, 2022-07-07:
cv32e40s_csr_template.yaml was updated in #1328, this issue serves as a remainder to generate a new test-section for the csr_access_test. (Test is automatically generated with some minor hand-modified sections)

silabs-hfegran, 2022-07-11:
csr_access_test script (from riscv-dv) does not support our interpretation of WARL - all bits in WARL-defined register fields are assumed fully RW by the test generator.

One example:
write: 0x5a5a_5a5a to mstatus, 
reads back:  0x0020_0080, while test generator expects 0x0020_1880 to be read back due to the WARL definition.

silabs-hfegran, 2024-01-11:
This is resolved

closed 2024-01-11 by silabs-hfegran

MikeOpenHWGroup, 2024-01-11:
Hi @silabs-hfegran, how was this issue resolved?  Was a modification of the CSR test-generator required?  If so, can you point to the associated PR?  Thanks.

silabs-hfegran, 2024-01-12:
https://github.com/silabs-hfegran/riscv-dv/commit/0b707d80452d2f7dcf68012645fc4ae6bbae2f1c (This is also present on upstream riscv-dv), there is a "legalize-warl"-functionality added to the script, that enables one to define the warl-behavior of a register with python in the yaml definition file.