openhwfoundation/core-v-verif #1341: vptool installation dependancies and execution

eroom1966, 2022-07-18:
@JeanRochCoulon 
I saw a new tool has been created called vptool.
I followed the instructions in VPTOOL-readme.txt for installation of packages, then whent to the vptool-example directory
trying to run 'runme.sh' results in

> 
> openhw/core-v-verif/tools/vptool/vptool-example$ ./runme.sh 
> export PLATFORM_TOP_DIR=/home/moore/git/openhw/core-v-verif/tools/vptool/vptool-example/example-database
> export PROJECT_NAME="VPTOOL example project"
> python3 /home/moore/git/openhw/core-v-verif/tools/vptool/vptool-example/../vptool/vp.py -t winxpblue
> Traceback (most recent call last):
>   File "/home/moore/git/openhw/core-v-verif/tools/vptool/vptool-example/../vptool/vp.py", line 21, in <module>
>     from typing import OrderedDict
> ImportError: cannot import name 'OrderedDict'

Is this unexpected ?
rather than debug this myself, I thought it would be useful to pass back the issue to update the readme

just in case there are OS issues, my OS is 
NAME="Ubuntu"
VERSION="18.04.3 LTS (Bionic Beaver)"

JeanRochCoulon, 2022-07-18:
Hello @eroom1966, Happy to have users to improve the tool and the related documentation. @zchamski is the owner of the tool, I will come back from holidays on August 8th. I think he will be please to answer your questions. Cheers

Imperas, 2022-07-19:
@JeanRochCoulon its not help with the documentation we need - we just cant get the technology you have checked in to run at all. Simon

eroom1966, 2022-08-23:
@JeanRochCoulon @zchamski any update on this ?

zchamski, 2022-08-23:
Hi @eroom1966, I got my hands on an Ubuntu 18.04 machine, update/fix coming soon

zchamski, 2022-08-30:
Hi @eroom1966 , did you have a chance/moment to test the fix that @MikeOpenHWGroup merged into master at c1b43bc?

Imperas, 2022-08-30:
I should have some time this week - thx

closed 2023-02-21 by eroom1966