openhwfoundation/core-v-verif #1343: [TASK] Verification configuration

JeanRochCoulon, 2022-07-19:
### Is there an existing core-v-verif task for this?

- [X] I have searched the existing task issues

### Task Description

 We have to implement a process to control the testbench memory model addressing, test-program compile flags and the core configuation at compile and run-time.

Please read https://github.com/openhwgroup/core-v-verif/pull/1342 for more information

### Description of Done

Support the smae features than CV32E4 verification. @MikeOpenHWGroup could help validating the complation of the task.

jquevremont, 2022-09-16:
Configurations defined. Not yet implemented.

jquevremont, 2023-01-20:
Need to talk with @MikeOpenHWGroup to check the implementation is OK.

jquevremont, 2023-03-14:
We are satisfied with the current flow for 1 CVA6 configuration.

closed 2023-03-14 by jquevremont