openhwfoundation/core-v-verif #1354: xmsim: *E,ASRTST, uvmt_cv32e40s_pmp_assert.sv

silabs-kun, 2022-07-28:


Silabs-ArjanB, 2022-07-28:
Transferred to core-v-verif

MikeOpenHWGroup, 2022-07-28:
This issue was moved and then closed.  It looks like the closure was not intended.  @silabs-kun is this issue now resolved?

silabs-robin, 2022-07-28:
Hi Mike. The issue is not resolved. It can be assigned to me, as I plan to look into it.

silabs-kun, 2022-08-02:
Hi Mike, problem solved by changing ```always @*``` to ```always_comb``` in uvmt_cv32e40s_pmp_assert.sv.

MikeOpenHWGroup, 2022-08-02:
Hi @silabs-kun, thanks for drilling into this and getting it resolved.  A couple of comments/questons:

1. Was this issue in code that was in the central (openhw) repo?  If the problem only existed in your fork, it is not necessary to create an issue.  If this issue _was_ related to code in the openhw repo, please refer to the pull-request that resolved this issue.
2. Do you understand why `always_comb` resolved the issue?  [This](https://www.verilogpro.com/systemverilog-always_comb-always_ff/) article may be useful.  The distinction is subtle.

silabs-kun, 2022-08-02:
> Hi @silabs-kun, thanks for drilling into this and getting it resolved. A couple of comments/questons:
>
> 1. Was this issue in code that was in the central (openhw) repo?  If the problem only existed [...]

1. I assume this exist in the central (openhw)repo as well since my fork is directly cloned from there.
2. Thank you very much for the info, I'll look into it.

silabs-robin, 2022-08-03:
1. The problem is in the central repo, yes. It manifested as an assertion error in simulation that couldn't be replicated in formal (so technically, the formal tool is possibly violating the LRM). There is no pull request that fixes the problem yet, so maybe we closed this ticket too early(!), but I also have the same fixes in a completely different branch that I am working on so that gives some additional assurance that it will arrive. We could re-open this issue until one of those are merged?

2. The issue is that the pmp assertions have logic that uses functions, but due LRM subtleties (see picture below) this is not included in the implicit sensitivity list.

[image]

MikeOpenHWGroup, 2022-08-03:
Thanks for this @silabs-robin, very clear.  In response to your question:

> There is no pull request that fixes the problem yet, so maybe we closed this ticket too early(!), but I also have the same fixes in a completely different branch that I am working on so that gives [...]

Yes, I think we should re-open this issue.

silabs-robin, 2022-11-18:
This issue is fixed. The PMP assertion files now use `always_comb`. Please close the issue.

closed 2022-11-18 by silabs-hfegran