openhwfoundation/core-v-verif #1366: ISS: dcsr.prv bit-field does not update in the rvvi

silabs-anvesten, 2022-08-10:
### Comment
Writing to the dcsr.prv bit-field results in a mismatch between the rvfi and rvvi. The ISS seems to not update the value while the core does.

### Printed failure message
UVM_ERROR @ 4287.300 ns : uvme_cv32e40s_core_sb.sv(383) uvm_test_top.env.core_sb [CORESB] CSR Mismatch, order: 294, pc: 0x1a11085e, csr: dcsr, rvfi = 0x400000c0, rvvi = 0x400000c3, mask = 0xffffffff
UVM_ERROR @ 4293.300 ns : uvme_cv32e40s_core_sb.sv(383) uvm_test_top.env.core_sb [CORESB] CSR Mismatch, order: 295, pc: 0x1a110862, csr: dcsr, rvfi = 0x400000c0, rvvi = 0x400000c3, mask = 0xffffffff
UVM_ERROR @ 4302.300 ns : uvme_cv32e40s_core_sb.sv(383) uvm_test_top.env.core_sb [CORESB] CSR Mismatch, order: 296, pc: 0x1a110864, csr: dcsr, rvfi = 0x400000c0, rvvi = 0x400000c3, mask = 0xffffffff
UVM_ERROR @ 4314.300 ns : uvme_cv32e40s_core_sb.sv(383) uvm_test_top.env.core_sb [CORESB] CSR Mismatch, order: 297, pc: 0x1a110866, csr: dcsr, rvfi = 0x400000c0, rvvi = 0x400000c3, mask = 0xffffffff
UVM_ERROR @ 4326.300 ns : uvme_cv32e40s_core_sb.sv(383) uvm_test_top.env.core_sb [CORESB] CSR Mismatch, order: 298, pc: 0x1a11086a, csr: dcsr, rvfi = 0x400000c0, rvvi = 0x400000c3, mask = 0xffffffff

### Steps to Reproduce
Checkout the DebugBug in my fork:
     https://github.com/silabs-anvesten/core-v-verif/tree/DebugBug
git hash: 
     c8ed2efc63ad8f31cf90603f19fcd35c1dd45247

make command: 
    make test TEST=debug_priv_test CFG=pmp

silabs-mateilga, 2022-08-10:
@eroom1966

eroom1966, 2022-08-11:
I tried to reproduce this issue in the following way (first running the sanity test)

git clone https://github.com/silabs-anvesten/core-v-verif
cd core-v-verif
git checkout c8ed2ef
make clean_all
make sanity CV_CORE=cv32e40x

But I get an error
> UVM_INFO @ 0.900 ns : uvmt_cv32e40x_tb.sv(705) reporter [OVPSIM] Start benchmarking OVPSIM initialization
> Fatal (UNSVAR) Variant CV32E40X_V0.0.0 is unsupported
> Info Exiting

Please advise where I made an error in my steps to reproduce

eroom1966, 2022-08-11:
Looks like an error in this file 
./cv32e40x/tb/uvmt/uvmt_cv32e40x_iss_wrap.sv:   CPU #(.ID(ID), .VARIANT("**CV32E40X_V0.0.0**")) cpu(bus, io);

eroom1966, 2022-08-11:
I also tried 

git checkout DebugBug
make test TEST=debug_priv_test CFG=pmp

And I got a different error

> make test TEST=debug_priv_test CFG=pmp
> CRITICAL:yaml2make:Could not find [test.yaml] in any directories:
> CRITICAL:yaml2make:/scratch/moore/git/TestCase/core-v-verif/cv32e40x/tests/programs/corev-dv/debug_priv_test/test.yaml
> CRITICAL:yaml2make:/scratch/moore/git/TestCase/core-v-verif/cv32e40x/tests/programs/embench/debug_priv_test/test.yaml
> CRITICAL:yaml2make:/scratch/moore/git/TestCase/core-v-verif/cv32e40x/tests/programs/custom/debug_priv_test/test.yaml
> /scratch/moore/git/TestCase/core-v-verif/mk/Common.mk:223: *** ERROR Could not find test.yaml for test: debug_priv_test. Stop.

silabs-anvesten, 2022-08-11:
My bad, I should have included this is in the original info, this is for the cv32e40s core, not the -40x

eroom1966, 2022-08-11:
I can see the issue here, there is some common code in the model between the E40P, E40X and E40S
this field is defined in the online docs as follows

E40P - 0x3 RO
E40X - 0x3 WARL
E40S - 0x0, 0x3 WARL

Thx
Lee

eroom1966, 2023-01-27:
returning to this issue, I have run the testcase on the 0.8.0 PR https://github.com/openhwgroup/core-v-verif/pull/1599 I have pending

make test TEST=debug_priv_test CFG=pmp

The result is as follows
firstly it tries to set a non-existant override Zcmb - this is not present on the E40S

Having corrected this, we then see a difference on a write to pmpcfg0 register, here is a short transcript

```
Info 414: 'refRoot/cpu', 0x0000000000000230(setup_pmp+8): Machine 42bd     c.li    x5,15
Info   MEMX 0x230 0x230 2 42bd
Info   x5 ffffffff -> 0000000f
Info 415: 'refRoot/cpu', 0x0000000000000232(setup_pmp+a): Machine 3a029073 csrw    pmpcfg0,x5
Info   MEMX 0x232 0x232 2 9073
Info   MEMX 0x234 0x234 2 3a02
Error (IDV) CSR register value mismatch (HartId:0, PC:0x00000232 setup_pmp+10):
Info (IDV)  0> CSR 3a0 (pmpcfg0)
Info (IDV)   . dut:0x0000000f
Info (IDV)   . ref:0xffffffff
UVM_ERROR @ 4267.300 ns : rvvi-pkg.sv(94) reporter [] uvmt_cv32e40s_tb.imperas_dv.trace2api.state_compare @ 4267.000 ns: MISMATCH
```

The pmpcfg0 is initially 0xFFFFFFFF, this means that the L bit is set - meaning the register is locked
So this looks like an error that the RTL is allowed to write the pmpcfg0 register when it is locked

silabs-robin, 2023-02-01:
Hi Lee. I want to have a look at this, but we currently have a problem with the PMP CSRs' default values and overrides. Once we have sorted that out I will come back to this.

MikeOpenHWGroup, 2023-05-16:
Hi @silabs-anvesten, @silabs-robin this issue has "gone stale".  Is it still an issue or can it be closed?

MikeOpenHWGroup, 2023-09-01:
Hi @silabs-hfegran, this issue has "gone stale". Is it still an issue or can it be closed?

silabs-robin, 2023-09-04:
That test/config is not failing anymore.
Any we have had several ISS updates since.
So I'd say we close this.

closed 2023-09-04 by silabs-robin