openhwfoundation/core-v-verif #1368: Generation of VCD for cva6 using verilator

spidugu444, 2022-08-11:
Hi All,

I am trying to enable vcd on verilator for cva6 core, could anyone suggest where to enable vcd in core-v-verif environment.

Thanks in advance.

MikeOpenHWGroup, 2022-08-12:
@JeanRochCoulon, @Gchauvon, @ASintzoff, can either of you answer @spidugu444's question?

zchamski, 2022-08-29:
@MikeOpenHWGroup , @spidugu444 : this is currently work-in-progress at my side because the task is not exactly straightforward.  First shareable results could arrive as soon as the end of this week (Sep 2).

zchamski, 2022-09-16:
The current state of work is available as PR #1417 and its companion PR in CVA6 https://github.com/openhwgroup/cva6/pull/954.

The waveform support is operational on VCS and Verilator setups and is controlled by the same variables in each evnrionment (see the description of the PR at the top of this page.)

Please note that the ***usage of the `DEBUG` variable is subject to change*** (it is likely to become invisible to the user.)

zchamski, 2022-10-03:
The support is now available in the mainline.  The revised PRs #1427 and https://github.com/openhwgroup/cva6/pull/965 have been merged into their respective base branches:
* `core-v-verif`: branch `cva6/dev`, commit https://github.com/openhwgroup/core-v-verif/commit/8f9e8e03b50ee846e80de837f3bca21432715e84
* `cva6`: branch `master`, commit https://github.com/openhwgroup/cva6/commit/871be7c7943820ebe72c85c72f9af1164a50dc32.

MikeOpenHWGroup, 2022-10-03:
Hi @spidugu444, does #1427 satisfy your request?  If so, please close this issue.  Thanks.

cathales, 2023-05-02:
Hello @spidugu444 

Is it okay for you now ?

MikeOpenHWGroup, 2023-09-14:
Closing because the issue is deemed to be resolved and the originator (@spidugu444) has not responded in four months.

closed 2023-09-14 by MikeOpenHWGroup