openhwfoundation/core-v-verif #1369: Floating point multipication issue in RISCV cva6

spidugu444, 2022-08-11:
Hi All,

In CVA6 riscv-arch-test suite fmul tests are failing due to incorrect calculation as shown in below attached screenshot.

[image]

As shown in above in line 157 and 182 expected result is 0,as it  is multiplied with zero but in verilator it is giving value as 80000000 where as in spike it is calculating correct result as 0.

MikeOpenHWGroup, 2022-08-12:
Thanks for this issue @spidugu444.  I have assigned it to @JeanRochCoulon, the verification lead for the CVA6.  As far as I know, the CVA6 team is not planning to support floating point operations in the initial release of CVA6, so this bug may not be a priority.

FYI, the floating point IP that is used by the CVA6 (and CV32E40P), known as `fpnew`, has recently been transferred from the pulp-platform team to the OpenHW Group and re-branded as [cvfpu](https://github.com/openhwgroup/cvfpu).  The OpenHW Group is starting a complete verification of this IP now.

The first user of cvfpu will be the v2.0.0 release of CV32E40P.  It is my expectation that the CVA6 team will integrate the fully verified `cvfpu` into CVA6, but I do not know the timeline for this.  In the meantime, I will created an issue on the cvfpu repo to link to this issue.

JeanRochCoulon, 2022-08-16:
Thank for this feedback.
You are using Verilator, but do you have the bug when using VCS ?   If no, which Verilator version do you use?

spidugu444, 2022-08-17:
@JeanRochCoulon  fmul test is failed on VCS also, currently I am using verilator-4.110 version.