openhwfoundation/core-v-verif #1374: Mstatus bug in RISCV CVA6

AyoubJalali, 2022-08-18:
I have been using the RISCV-DV (I use VCS to compile and run the generator), to improve the Code coverage of the CVA6 Core processor, by generating tests from the RISCV-DV using a YAML description, and while trying to access the CVA6’ CSRs, After a failed test, comparing the output of CVA6 with the reference model (spike), I found a mismatch between CVA6 and ref model Spike, the mismatch was in the Mstatus register, is a MXLEN-bit read/write register (64 bits for my case), formatted as shown in Figure 3.7 in (The RISC-V Instruction Set Manual Volume II: Privileged Architecture). The mstatus register keeps track of and controls the hart’s current operating state.
 
The problem appears clearly in WPRI (Some whole read/write fields are reserved for future use. Software should ignore the values read from these fields, and should preserve the values held in these fields when writing values to other fields of the same register. For forward compatibility, implementations that do not furnish these fields must make them read-only zero. These fields are labeled WPRI in the register descriptions) field (Spike read-only zero, while CVA6 read a different value).
In some tests, we found a mismatch also in other fields, for example the UBE field, SPP and SBE.
So to reproduce the problem, I recommend use the RISCV-DV using the YAML description in cva6/dev branch, core-v-verif/cva6/sim/cva6_base_testlist.yaml (https://github.com/openhwgroup/core-v-verif/blob/cva6/dev/cva6/sim/cva6_base_testlist.yaml), adding to this list the description to reproduce the bug:  

- test: Mstatus_bug
  description: >
    Arithmetic instruction test, with CSR instructions
  gen_opts: >
    +instr_cnt=500
    +num_of_sub_program=0
    +directed_instr_0=riscv_int_numeric_corner_stream,5
    +no_fence=1
    +no_data_page=1
    +no_branch_jump=1
    +boot_mode=m
    +no_csr_instr=0
    +randomize_csr=1
    +enable_illegal_csr_instruction=1
  iterations: 2
  gen_test: riscv_instr_base_test
  rtl_test: core_base_test

After that run the RISCV-DV:

python3 cva6.py --testlist=cva6_base_testlist.yaml --test Mstatus_bug --iss_yaml cva6.yaml --target cv64a6_imafdc_sv39 --iss=vcs-uvm,spike -i 10  --iss_timeout 300 (run it in core-v-verif/cva6/sim)

10 iterations can be enough to found t

seriesAB, 2023-06-20:
How do i assign this issue to myself?

MikeOpenHWGroup, 2023-06-20:
Hi @seriesAB, thanks for your interest in CVA6.  This is _not_ a good first issue and is not in any way related to #1404.  I would recommend focusing on that one first.  Thanks!