openhwfoundation/core-v-verif #1376: Documentation: rvfi_trap size declaration

eroom1966, 2022-08-19:


Silabs-ArjanB, 2022-08-19:
@eroom1966 I transferred it

Silabs-ArjanB, 2022-08-19:
@silabs-mateilga Could you please have a look at this?

silabs-mateilga, 2022-08-22:
I'll have a look at this today.

silabs-mateilga, 2022-08-22:
Fixed in PR [1378](https://github.com/openhwgroup/core-v-verif/pull/1378)

MikeOpenHWGroup, 2022-08-23:
Can this issue be closed?

eroom1966, 2022-08-23:
I took the PR and it looks like I have a width error generated - I posted on the PR

silabs-mateilga, 2022-08-23:
Hmm, this is weird, it passed ci_check before merge. I'll have a look at it again tomorrow.

eroom1966, 2022-08-23:
I think that is because it is only a warning - not an error

silabs-mateilga, 2022-08-24:
Put in the effort and fixed the core of the issue this time, by changing to the struct used in the core.

https://github.com/openhwgroup/core-v-verif/pull/1380

MikeOpenHWGroup, 2022-08-24:
Thanks @silabs-mateilga.  Back to you @eroom1966 - can this issue be closed?

eroom1966, 2022-08-24:
yes I tested this and no more warnings - thx

closed 2022-08-24 by eroom1966