openhwfoundation/core-v-verif #1382: Testbench adaptation

AEzzejjari, 2022-08-24:
Hello, @ASintzoff 
I have been improving code converged for AXI modules . I have executed tests that fail. These tests execute atomic operations. I found that the testbench does not support these operations, for that it's necessary to adapt it to make sure that the CVA6 respects the AXI protocol during the communication with the memory.

MikeOpenHWGroup, 2022-08-24:
Hi @AEzzejjari, thanks for your issue.  I agree with you and **strongly** recommend to the CVA6 team that this issue be resolved by either adopting (if we can find one) or developing an AXI UVM Agent similar to the UVM [OBI Agent](https://github.com/openhwgroup/core-v-verif/tree/master/lib/uvm_agents/uvma_obi_memory) used by the UVM testbenches for the CV32E4 cores.

This will be a big task, and it is worthwhile to capture it as a task on the CVA6 project board.

JeanRochCoulon, 2023-03-16:
@AEzzejjari   As Today the system memory which is used by UVM testbench is part of AXI agent. You became the best candidate to implement it inside !  @Gchauvon is maybe the best person to refer for this topics if you need more information on features to be implemented.