openhwfoundation/core-v-verif #1391: ELF vs HEX mismatch

silabs-hfegran, 2022-08-30:
## ELF vs HEX mismatch

### Type
* Discrepancy between how binutils and the ISS determines which regions that are to be loaded into memory
* Possibly a linker bug, causing incorrect program header in elf-file
* If linker behavior is correct (TBD), then parsing the program header table is not sufficient to determine what parts of the binary to be loaded into system memory for a bare metal system. (indicated by objcopy behavior outlined below)

### Background
ISS mismatches observed with attempts to read from memory 0x90f4 in a random simulations. The values being read by the ISS from the region 0x90xx appears to match the contents of the ELF header, while RTL reads zero (expected behavior)

Inspecting the program and section header tables (see below) indicates that extra space is allocated to the program header entry with the lowest address (+0x1000), with a start offset at 0x0 (start of file - which is always the elf header). Cutter disassembly also confirms that address 0x9000 is marked as loadable, and pointing to the header itself.
consequently, ISS parses the program headers in the elf file to determine which parts of the elf file to load into memory, including file offset 0x0-0xFFF into offset 0x9000-0x9FFF, prior to loading the actual code that starts at address 0xA000.

This behavior is only observed in cases where the binary does not start at address 0x0000_0000.

In core-v-verif, the binutils `<riscv-toolchain>_objcopy -O verilog` command is used to generate the memory image (hex file) used by verilog simulators; this hex is generated as expected and does not contain the elf header, thus there is a discrepancy between how objcopy and the ISS determines which regions to load into memory.

### Steps to Reproduce

* branch: cv32e40x/dev
* hash: bf064fc0b11b88bb6483ac10766f3ab02abee01f

* Generate source code and linker scripts:
`make comp_corev-dv gen_corev-dv TEST=corev_rand_fencei CV_CORE=cv32e40x CFG=pma_test_cfg_3 SIMULATOR=xrun USE_ISS=NO SEED=819033859 `

* Build and run (This performs the necessary step to generate the output binaries):
`make test TEST=corev_rand_fencei CV_CORE=cv32e40x CFG=pma_test_cfg_3 SIMULATOR=xrun USE_ISS=NO SEED=969359103 `
* This does not get flagged as an error with these steps, but visual inspection of the output files confirm that the issue is present

### Readelf output:

This snippet from the program headers is the culprit, "load 0x11000 bytes from elf file offset 0x0 into cpu memory at offset 0x9000", loading the elf header into bare metal cpu memory makes absolutely no sense, and is probably why objcopy discards it:
 |   Type           |  Offset   |  VirtAddr   |  PhysAddr   |  FileSiz |  MemSiz  |  Flg |  Align   |
 |---               |---        |---          |---          |---       |---       |---   |---     |
|   **LOAD**           | **0x000000** |  **0x00009000** |  **0x00009000** |  **0x11000** |  **0x11000** |  RW  |  0x1000   |

**Program Headers:**
 |   Type           |  Offset   |  VirtAddr   |  PhysAddr   |  FileSiz |  MemSiz  |  Flg |  Align   |
 |---               |---        |---          |---          |---       |---       |---   |---     |
 |   RISCV_ATTRIBUT |  0x15affe |  0x00000000 |  0x00000000 |  0x00044 |  0x00000 |  R   |  0x1   |
 |   **LOAD**           | **0x000000** |  **0x00009000** |  **0x00009000** |  **0x11000** |  **0x11000** |  RW  |  0x1000   |
 |   LOAD           |  0x011000 |  0x02000000 |  0x02000000 |  0x10000 |  0x10000 |  RW  |  0x1000   |
 |   LOAD           |  0x0210f0 |  0x100000f0 |  0x100000f0 |  0x10000 |  0x10000 |  RW  |  0x1000   |
 |   LOAD           |  0x031a54 |  0x13acaa54 |  0x13acaa54 |  0x10000 |  0x10000 |  RW  |  0x1000   |
 |   LOAD           |  0x042800 |  0x1a110800 |  0x1a110800 |  0x00802 |  0x00890 |  RWE |  0x1000   |
 |   LOAD           |  0x043008 |  0x23400008 |  0x23400008 |  0x10000 |  0x10000 |  RW  |  0x1000   |
 |   LOAD           |  0x054000 |  0x2a000000 |  0x2a000000 |  0x10000 |  0x10000 |  RW  |  0x1000   |
 |   LOAD           |  0x064ca8 |  0x3100fca8 |  0x3100fca8 |  0x10000 |  0x10000 |  RW  |  0x1000   |
 |   LOAD           |  0x075854 |  0x3420c854 |  0x3420c854 |  0x10000 |  0x10000 |  RW  |  0x1000   |
 |   LOAD           |  0x086000 |  0x3600a000 |  0x3600a000 |  0x10000 |  0x10000 |  RW  |  0x1000   |
 |   LOAD           |  0x096000 |  0x3ace0000 |  0x3ace0000 |  0x10000 |  0x10000 |  RW  |  0x1000   |
 |   LOAD           |  0x0a6000 |  0x48000000 |  0x48000000 |  0x00080 |  0x00080 |  R E |  0x1000   |
 |   LOAD           |  0x0a6080 |  0x48000080 |  0x48000080 |  0x32e88 |  0x11ff80 | RWE | 0x1000   |
 |   LOAD           |  0x000f10 |  0x48032f10 |  0x48032f10 |  0x00000 |  0xed0f0 |  RW  |  0x1000   |
 |   LOAD           |  0x0d8ffc |  0x4bfffffc |  0x4bfffffc |  0x10000 |  0x10000 |  RW  |  0x1000   |
 |   LOAD           |  0x0e8ffc |  0x4f99fffc |  0x4f99fffc |  0x10000 |  0x10000 |  RW  |  0x1000   |
 |   LOAD           |  0x0f9bfc |  0x5000abfc |  0x5000abfc |  0x010c8 |  0x010c8 |  RW  |  0x1000   |
 |   LOAD           |  0x0faffc |  0x52fffffc |  0x52fffffc |  0x10000 |  0x10000 |  RW  |  0x1000   |
 |   LOAD           |  0x10affc |  0x56fffffc |  0x56fffffc |  0x10000 |  0x10000 |  RW  |  0x1000   |
 |   LOAD           |  0x11affc |  0x63fffffc |  0x63fffffc |  0x10000 |  0x10000 |  RW  |  0x1000   |
 |   LOAD           |  0x12affc |  0x7ffffffc |  0x7ffffffc |  0x10000 |  0x10000 |  RW  |  0x1000   |
 |   LOAD           |  0x13affc |  0x8459fffc |  0x8459fffc |  0x10000 |  0x10000 |  RW  |  0x1000   |
 |   LOAD           |  0x14affc |  0xeffffffc |  0xeffffffc |  0x10000 |  0x10000 |  RW  |  0x1000   |

**Section to Segment mapping:**
 |  Segment | Sections...
 |--        |--
 |   00     | .riscv.attributes 
 |   01     | .region_28 
 |   02     | .region_26 
[... 80 lines cut ...]
| region_6         | 0x000000003420c854 | 0x0000000000120000 | axrwl
| region_7         | 0x000000004f99fffc | 0x0000000000120000 | axrwl
| region_8         | 0x000000003100fca8 | 0x0000000000120000 | axrwl
| region_9         | 0x000000005000abfc | 0x00000000000010c8 | axrwl
| region_10        | 0x0000000030001350 | 0x0000000000120000 | !l
| region_11        | 0x000000005000bcc8 | 0x0000000000120000 | !l
| region_12        | 0x000000002c5a3200 | 0x0000000000120000 | !l
| region_13        | 0x000000005140fffc | 0x0000000000120000 | !l
| region_14        | 0x000000002a000000 | 0x0000000000120000 | axrwl
| region_15        | 0x0000000052fffffc | 0x0000000000120000 | axrwl
| region_16        | 0x0000000023400008 | 0x0000000000120000 | axrwl
| region_17        | 0x0000000056fffffc | 0x0000000000120000 | axrwl
| region_18        | 0x0000000020000000 | 0x0000000000120000 | !l
| region_19        | 0x00000000600ffffc | 0x0000000000120000 | !l
| region_20        | 0x0000000013acaa54 | 0x0000000000120000 | axrwl
| region_21        | 0x0000000063fffffc | 0x0000000000120000 | axrwl
| region_22        | 0x00000000100000f0 | 0x0000000000120000 | axrwl
| region_23        | 0x000000007ffffffc | 0x0000000000120000 | axrwl
| region_24        | 0x0000000005000000 | 0x0000000000120000 | !l
| region_25        | 0x0000000082fffffc | 0x0000000000120000 | !l
| region_26        | 0x0000000002000000 | 0x0000000000120000 | axrwl
| region_27        | 0x000000008459fffc | 0x0000000000120000 | axrwl
| **region_28**        | **0x000000000000a000** | 0x0000000000120000 | axrwl
| region_29        | 0x00000000effffffc | 0x0000000000120000 | axrwl
| *default*        | 0x0000000000000000 | 0xffffffffffffffff

jeremybennett, 2022-08-30:
hi @silabs-hfegran Would you be able to create a smaller reproducer that doesn't depend on EDA tooling, and I will take a look.

silabs-hfegran, 2022-08-31:
> hi @silabs-hfegran Would you be able to create a smaller reproducer that doesn't depend on EDA tooling, and I will take a look.

Thanks for looking into this, @jeremybennett, I will try to create a small, self-contained example

silabs-hfegran, 2022-09-01:
https://github.com/silabs-hfegran/debug/tree/elf_issue_1391

silabs-hfegran, 2023-06-23:
This issue can be worked around with the `-nmagic` flag to the linker.

closed 2023-06-23 by silabs-hfegran