openhwfoundation/core-v-verif #1404: Unify Interrupt and Debug Assertion in E-core testbenches

MikeOpenHWGroup, 2022-09-05:
### Task Outcome
As shown in the Figure below, in core-v-verif there are two methods to assert interrupts and debug requests to the core.  `Synchronous interrupts and debug requests` are invoked by the test-program which writes to a virtual peripheral supported by the memory model.  `Asynchronous interrupts and debug requests` can be triggered by a UVM Agent (not shown in the Figure).

This works well, but it it means that synchronous events triggered by the virtual peripherals are not visible to the agent and therefore are not recorded in the associated transaction log file (e.g. uvm_test_top.env.debug_agent.trn.log).  Only the asynchronous events created by the interrupt and debug agents appear in the logfiles.

When this task is complete, the `uvm_test_top.env.debug_agent.trn.log` and `uvm_test_top.env.interrupt_agent.trn.log` files will reflect both asynchronous and synchronous events.

### Background information
A description of the virtual peripherals and their operation is [here](https://docs.openhwgroup.org/projects/core-v-verif/en/latest/test_programs.html#virtual-peripherals).

### Location Information
The debug and interrupt agents are found under the `lib/uvm_agent` directory in core-v-verif.

### Completion Criteria
All existing tests, specifically those which assert interrupts and/or debug results should run as before.  The only difference is that the logfiles produced by the debug and interrupt agent will now include interrupts/debug event initiated by both the virtual peripherals and sequence items.

### Additional context
A schematic illustrating how synchronous and asynchronous interrupts and debug requests are "or-ed" together in the current E40X testbench is shown below.
[image]

seriesAB, 2023-06-20:
How do i assign this issue to myself?

MikeOpenHWGroup, 2023-06-20:
Hi @seriesAB, thanks for your interest in CORE-V-VERIF!  Assigning issues can only be performed by OpenHW Committers (reach out to me via email if you want to know more about that).  I've gone ahead and assigned this issue to you.

Please read [CONTRIBUTING](https://github.com/openhwgroup/core-v-verif/blob/master/CONTRIBUTING.md) before issuing any pull-request.  Thanks!

seriesAB, 2023-06-20:
Thank You ! i will read those and get back to you!

seriesAB, 2023-06-21:
Could you give me the use case - input/output for #1404

seriesAB, 2023-06-21:
Could you give me 5 use cases - input/output for [#1404](https://github.com/openhwgroup/core-v-verif/issues/1404)

seriesAB, 2023-06-21:
Could you give me 5 use cases - input/output for [#1404]

seriesAB, 2023-06-21:
How do test this code base after making the changes to make sure nothing is breaking

MikeOpenHWGroup, 2023-06-21:
> I am going to import uvm_agent into the virtual peripheral and log from there - is that okay?

The virtual peripherals are all implemented as UVM virtual sequences so importing the agent is not the right approach here.  There is no need to modify the interrupt and debug agents, you just need to ensure that the monitors in those agents can "see" both the `uvma` and `vp` instances of the debug interfaces in the testbench.

MikeOpenHWGroup, 2023-06-21:
> How do i test the code base after making changes?

I would suggest three tests:
```
$ make test TEST=hello-world USE_ISS=NO SIMULATOR=<vcs|xrun|vsim>
$ make test TEST=debug_test USE_ISS=NO SIMULATOR=<vcs|xrun|vsim>
$ make test TEST=interrupt_test USE_ISS=NO SIMULATOR=<vcs|xrun|vsim>
```
Get these working in your environment before making any changes.

seriesAB, 2023-06-21:
Okay ,how do i do a full test of the code base using regression yaml ?

seriesAB, 2023-06-21:
Okay thank You