openhwfoundation/core-v-verif #1406: e40x/s - Lump assert files together

silabs-robin, 2022-09-06:
From discussion in https://github.com/openhwgroup/core-v-verif/pull/1399#discussion_r962960940

--------

We have many assertion files and they are all just dumped into the `tb/uvmt/` directory.
Let's create `tb/asserts/` or something similar and have the files more cleanly organized that way.

Additionally, remove the `core/` stuff from the uvmt ".flist" file.

Consider also, whether we might want a separate ".flist" file for the assertion files (to help share code between sim/fv).

silabs-robin, 2022-09-06:
_(The inference is "lumped > dumped")_

MikeOpenHWGroup, 2022-09-06:
Thanks for creating this issue @silabs-robin.  I think it will be of interest to the E40P team as well (@pascalgouedo, @YoannPruvost).  I like the idea of moving all assertion files from `<core>/tb/uvmt/uvmt_<core>_*_assert.sv` to `<core>/tb/assert/<core>_*_assert.sv`.

Having a separate manifest (flist) is also a good idea.  Hopefully we can create a unified one that can be used for both simulation and formal verification.

The `<core>/tb/assert` directory should also have a README listing the purpose of each file and which FV tool(s) have been used with the assertions.