openhwfoundation/core-v-verif #1408: Illegal DefParam in isacov

MikeOpenHWGroup, 2022-09-06:
### Type
This is a compile-time error that was flagged by Metrics DSIM.  It seems that Synopsys VCS accepts this code construct.

### Description
The issue is on [line 41 of uvma_rvfi_instr_mon.sv](https://github.com/openhwgroup/core-v-verif/blob/07b23ddad943efc9b19c7d8586d5f7b80e9bd8c5/lib/uvm_agents/uvma_rvfi/uvma_rvfi_instr_mon.sv#L41).  DSIM reports that:
```
A name of the form uvma_rvfi_csr_seq_item_c::XLEN where uvma_rvfi_csr_seq_item_c is    
the default specialization of a parameterized class                                    
is not valid.  Try uvma_rvfi_csr_seq_item_c#()::XLEN instead.
```
Making the suggested change resolves the issue.

### Additional Context
This issue was found on the `cva6/dev` branch of this repo.

JeanRochCoulon, 2022-09-06:
@ASintzoff, could you have a look at this issue ?

AyoubJalali, 2022-09-06:
@MikeOpenHWGroup I just create a PR for that, https://github.com/openhwgroup/core-v-verif/pull/1410
check if it work for Metrics DSIM (it work for synopsys VCS).

JeanRochCoulon, 2022-09-06:
Great @AyoubJalali, let's see if it works !

MikeOpenHWGroup, 2022-09-06:
Works for me.  Thanks for the quick turn-around @AyoubJalali.

closed 2022-09-06 by MikeOpenHWGroup