openhwfoundation/core-v-verif #1425: ISACOV: Incorrect length for rs1 & rs2 & rd in some compressed instructions

AyoubJalali, 2022-09-26:
Hello, @silabs-robin 
I notice in ### https://github.com/openhwgroup/core-v-verif/blob/master/lib/uvm_agents/uvma_isacov/cov/uvma_isacov_cov_model.sv there is some incorrect coverpoint declaration for some **C_TYPE** like **(CIW, CL, CS, CB)**, ### https://github.com/openhwgroup/core-v-verif/blob/05ba4eb0cf6296bf35f487fbe2ab01e532d0fe26/lib/uvm_agents/uvma_isacov/cov/uvma_isacov_cov_model.sv#L1293, so in this example we create a coverpoint for a 5 bits register (32 auto bins), and we know that for CB_TYPE the rs1 is only a 3 bits (8 auto bins), so that affect the real total value of the coverage.
**So for now a had a simple idea to correct that, instead of create a coverpoint for rs1, we can create it just for the first 3 bits, rs1[2:0],**if what i'm saying correct, should be more simple to add a more variable for the C extension, then connect it.

silabs-robin, 2022-09-28:
Hi.

I had to investigate a bit because I don't remember what all these things mean (and we weren't clever enough in the past to add descriptive names or comments).

> there is some incorrect coverpoint declaration for some C_TYPE like (CIW, CL, CS, CB)

Btw, we used "Spike" (aka "riscv-isa-sim") at the time because we had more faith in the decoding of a tried and true riscv implementation than in writing a new decoder for isacov. (We have actually had some discussion recently debating whether to write a simple decoder ourselves, to avoid dpi_dasm and spike updates etc, wrt e.g. Zc.)

I see in the monitor that (for RVC) rs1 is [fetched using dasm_rvc_rs1](https://github.com/openhwgroup/core-v-verif/blob/05ba4eb0cf6296bf35f487fbe2ab01e532d0fe26/lib/uvm_agents/uvma_isacov/uvma_isacov_mon.sv#L166).
And that ultimately links to the [decoding in Spike](https://github.com/riscv-software-src/riscv-isa-sim/blob/0654e4f75a80e43bdebed98be6af86c57cf1a26f/riscv/decode.h#L114).

IMO, the initial RVC support in ISACOV was just wrong.
Here is the part where the rs1 assignments etc were added https://github.com/openhwgroup/core-v-verif/commit/71ba0add772c46611ecfe37da702bd3d74306d15.
Isn't also the `rd'` of CIW sampled wrong?

> i'm not sure if we should take the LSB 3 bits, or MSB

The Spike decoding looks like `uint64_t rvc_rs1s() { return 8 + x(7, 3); }`, so that should be MSBs. I think that might be reflected in the name `.rs1` (since it is not `.rs1_primed` or anything similar, it is the actual `x*`).

> should be more simple to add a more variable for the C extension, then connect it.

I think I might prefer this idea the most.
When the variable is just called `.rs1` it is difficult to read whether this is MSB or LSB or if it even corresponds to the "primed" RVC version etc.
Ideally the name should signify its meaning such that we read it and think "ok, this is precisely `rd'/rs1'` being used in CA" etc.
But this is just my initial reaction, and I have not thought this through 110%, so I might have overlooked any potential implications of this (though I expect not).
_(Actually, in that commit I linked to above, I see that I had previously used `.c_rs1p` and `.c_rdp` etc, but someone else removed it.)_

AyoubJalali, 2022-09-28:
hello,
I prefer also to simply decode instructions, it difficult a little bit but it's simple to read instead of the **dpi_dasm**.
Yes also for CIW, i just give an example.
I would like to have a discuss later to see how we can re-implement C extension in ISACOV, for now i'll let this for now.
Thanks you for your respond.

MikeOpenHWGroup, 2022-09-28:
>  I prefer also to simply decode instructions, it difficult a little bit but it's simple to read instead of the dpi_dasm.

When ISACOV was in early development we considered this, and in the long-term I would very much like to have a decoder built into ISACOV (or as a stand-alone component used by ISACOV).  I estimate that this would be at least two-engineering-months of effort, so at the time we (myself and Steve Richmond of SiLabs) decided to use dpi_dasm.

If you would be interesting in helping OpenHW develop a decoder for this application, I will try to engage other members to contribute - there seems to be interest in this at the moment.

AyoubJalali, 2022-09-28:
> If you would be interesting in helping OpenHW develop a decoder for this application, I will try to engage other members to contribute - there seems to be interest in this at the moment.
Yes I would like

MikeOpenHWGroup, 2022-09-28:
Fantastic!  That would be a great contribution. Let's discuss that outside of this GitHub issue.

commit 789441a8 (2023-04-30): Merge pull request #1847 from AyoubJalali/fix/isacov

ISACOV : Fix #1425
commit b6675061 (2023-04-30): ISACOV : Fix #1425
closed 2023-05-26 by JeanRochCoulon