openhwfoundation/core-v-verif #1448: ISS in some circumstances flags cm.push as illegal instructions

silabs-hfegran, 2022-10-17:
ISS flags cm.push instruction as illegal instruction

It appears that if certain (unclear) conditions are met, the ISS flags cm.push instructions as illegal instructions.
While it is not entirely clear to me why this happens, it appears to be related to some kind of memory/stack corruption of sorts.

What is the expected response from the ISS in case of a `cm.push` instruction that pushes/allocates memory outside valid stack  boundaries? 

The code and log below executes a randomly generated test, that includes cm.push/cm.pop pairs at random intervals, with some arbitrary code inmixed. The instruction `0xb87e` ( `push {ra, s0-s2}, -64` ), is flagged as invalid at the end of the program execution log. What puzzles me, is that the exact same instruction word also appears at lines 7750, 6406 and 4286 in the log, where it is executed successfully. Tampering with stack size constraints that riscv-dv uses for its subprograms seems to correlate with the frequency of the error occurring; i.e. small stack -> first push instruction fails, large stack -> quite a few push instructions execute successfully until one suddenly fails. There is no memory protection scheme involved with this particular test. 

Code/binary that exhibits this behavior is located at the following github repository/branch:
<Placeholder>

eroom1966, 2022-10-17:
@silabs-hfegran 
Hi Henrik, 
Can you please give me the exact set of commands/steps to reproduce locally ?

thx
Lee

eroom1966, 2022-10-17:
OK, let me take another look

Thx
Lee

eroom1966, 2022-10-17:
@silabs-hfegran 
Can you double check - I do not see an elf file ?

eroom1966, 2022-10-17:
@silabs-hfegran 
You have selected compress_version 0.70.1
Also take a look at the overrides, it seems you are applying conflicting overrides - search the log for 
root/cpu/Compressed_Extension

eroom1966, 2022-10-17:
@silabs-hfegran 
is this the E40X ?
There were a number of fixes to the 0.70.1 spec, one of them I am pretty sure is related to the change related to stack pointer alignment.
Are you able to run this with the new model under the E40X/ImperasDV

eroom1966, 2022-10-17:
I saw this in the changelog for the model for version 0.70.1

>   - push/pop instructions no longer require the stack pointer to be aligned as was the case in previous specification versions.

eroom1966, 2022-10-17:
lets get this testcase into the new cv32e40x/dev environment, pretty sure it will be correct.

silabs-hfegran, 2022-10-18:
@eroom1966 Thanks for your answers! This was indeed on the 40S without the latest updates, thus I had no more recent option than 0.70.1 - I (falsly, I guess) assumed that since there were no changes to this instruction from that version till now that it should work as expected. I will run this test on the 40X as soon as I get the necessary functionality in on the 40S.

silabs-hfegran, 2022-10-18:
Apologies for the missing binaries, .gitignore was playing tricks on me - they are in now, but I guess there is no point in spending time on this until testing with the latest version.

eroom1966, 2022-11-04:
I have a fix for this, I can commit after a fix to #1472

eroom1966, 2022-11-08:
@silabs-hfegran are you able to re-test this with the latest commits ?

silabs-hfegran, 2022-11-18:
This appears to have been resolved, thanks! :)

closed 2022-11-18 by silabs-hfegran