openhwfoundation/core-v-verif #1453: ISACOV : wrong arguments for the toggle macro

AyoubJalali, 2022-10-25:
Hi, 
I think we have another problem related to difference between get_field_imm & get_data_imm, because while looking for some missing coverage, I notice that in some instructions we're trying to toggle a 6 bits imm using a macro to toggle a 8 bits imm, here lengths did not match and let us with some uncover bins, that's because in the macros we passed the get_filed_imm (return always 6 bits length) instead of get_data_imm (add some bits to the real length of imm, which i don't know why).
So I think we should modify some macros to match the length, because if we want to passed the get_data_imm, we're going to fall also in some uncover bins, because we add 0 bits, so we gonna have uncover bins related to 1 bits.
@silabs-robin what do you think here ?

silabs-robin, 2022-10-27:
I think this is one of the points on my list https://github.com/openhwgroup/core-v-verif/issues/1076, _"Fix indices of toggle macros (the "imm"s are now aligned downto 0)"_.

Using `get_data_imm` doesn't make sense, right? Because it will in many cases just append a zero to a number, so that zero cannot get toggled.
IIRC, I switched the coverpoints to using `get_field_imm` before, and I started fixing the toggles _(but I didn't get very far because I got directed to working on other tasks)_.

> So I think we should modify some macros to match the length

I think I very much agree with what you wrote.

AyoubJalali, 2022-10-27:
> Using `get_data_imm` doesn't make sense, right?

Yes , I think less we use  get_data_imm , better for coverage

closed 2024-07-10 by AyoubJalali