openhwfoundation/core-v-verif #1469: ISS default mcountinhibit value incorrect

silabs-hfegran, 2022-11-03:
## ISS mcountinhibit reset value incorrect

### Steps to Reproduce
cv32e40s/dev, hash https://github.com/openhwgroup/core-v-verif/commit/5db0790e98e5e022c7e51f14b9f40fdcc74564ca
make test TEST=hello-world CV_CORE=cv32e40s CFG=default USE_ISS=NO

Comment out lines 724, 725 in core-v-verif/cv32e40s/tb/uvmt/uvmt_cv32e40s_imperas_dv_wrap.sv (This is a temporary workaround for this issue)

### Additional context
Reset value of mcountinhibit is 0x0000_0004 in the ISS, and we are unable to override this with 
` --override cpu/extension/mcountinhibit_reset=5`
 `--override cpu/noinhibit_mask=0xfffffffd`

Spec v0.6.0 requires reset value to be set to 0x0000_0005, ISS traps on first instruction, expecting mcountinhibit to be 0x0000_0004
[image]

showoverrides shows that we have overridden these parameters, but there appears to be no effect: 
[image]
[image]

If submitting a fix for this issue, please remove the two lines disabling csr checks for this, lines 724/725 in aforementioned hash: 
[image]

eroom1966, 2022-11-04:
in order to make progress on this, I have a dependency on resolving #1472
Please re-assign once this is complete

eroom1966, 2022-11-04:
Apologies RTFM - just seen your comment about removing lines

closed 2022-11-17 by silabs-hfegran