openhwfoundation/core-v-verif #1470: Imperas ISS: Disable "Invalid instruction-warnings" to stdout possible?

silabs-hfegran, 2022-11-03:
Running invalid instruction tests, it could be very beneficial to suppress ISS' invalid instruction warnings, is there a way to do this? 

A log file with 100k 
`Illegal instruction (core 0) at PC xxxxxxx:`
`Warning (RISCV_UDEC) CPU 'refRoot/cpu' 0x0002633a d055c2eb undef: Undecoded instruction`
is not very useful, and the frequent writes to log files might incur a performance penalty

eroom1966, 2022-11-04:
I think this message is produced by the testbench, not the ISS
` Illegal instruction (core 0) at PC xxxxxxx:`

This message 
` Warning (RISCV_UDEC) CPU 'refRoot/cpu' 0x0002633a d055c2eb undef: Undecoded instruction`
Can be filtered by using the -excludem command, this is documented in the Control File User Guide

you should be able to add 
-excludem RISCV_UDEC

I also found a number of occasions where applications attempt to access non-=existant CSR's you can also exclude these
--excludem RISCV_CSR_UNIMP

MikeOpenHWGroup, 2022-11-04:
The "Illegal instruction" message comes from the "core_log" file in the CV32E40S ([link](https://github.com/openhwgroup/cv32e40s/blob/68a94befb5771e8b29ce445cfcf2de6de5877cb4/bhv/cv32e40s_core_log.sv#L58)).  This has been there since the RI5CY core days and I have been unsuccessful getting rid of it.

silabs-hfegran, 2022-11-07:
Thanks for clearing that up, I will create a PR to remove (or conditionalize) this

closed 2022-11-18 by silabs-hfegran